gqrx-sdr/gqrx #931: install dependencies for MacOS

ulrichloose, 2021-01-06:
Wouldn't it make sense to remove the "install dependencies" for MacOS during the checks? Depending on the temperature, mood, time of day, etc., mistakes can happen with homebrew. Basically these dependencies do not belong to gqrx but should be the responsibility of the user. Some compile themselves, others use homebrew and still other users prefer Macports ...

The constant false error messages when pushing are quite annoying.

argilo, 2021-01-06:
> Wouldn't it make sense to remove the "install dependencies" for MacOS during the checks?

No, it wouldn't. Gqrx cannot be built without its dependencies.

argilo, 2021-01-06:
To clarify: the dependencies need to be installed one way or another in the GitHub Actions builds. That doesn't necessarily need to be with Homebrew. Macports might be a possibility. Building the dependencies from source could also work, but it would be very slow to build GNU Radio.

closed 2021-01-06 by argilo

ulrichloose, 2021-01-08:
Maybe I just expressed myself a little misleadingly...

For a few days now I haven't been able to get rid of any pull request because the depency test aborts with error 1.

==> Pouring gcc@8-8.4.0_2.catalina.bottle.tar.gz
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into / usr / local
Could not symlink bin / c ++ - 8
Target / usr / local / bin / c ++ - 8
is a symlink belonging to gcc @ 8. You can unlink it:
brew unlink gcc @ 8
brew link --overwrite gcc @ 8
To list all files that would be deleted:
brew link --overwrite --dry-run gcc @ 8

and

==> Pouring gcc@9-9.3.0_1.catalina.bottle.tar.gz

Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into / usr / local
Could not symlink bin / c ++ - 9
Target / usr / local / bin / c ++ - 9
is a symlink belonging to gcc @ 9. You can unlink it:
brew unlink gcc @ 9
To force the link and overwrite all conflicting files:
brew link --overwrite gcc @ 9
To list all files that would be deleted:
brew link --overwrite --dry-run gcc @ 9

I admit - I have no clue about Github Actions, but for me the error message indicates that both GCC versions are installed without errors, but cannot be published in usr / local / bin via symlink. Because - that's my guess - there is already a link. I suspect that Github Actions is based on a virgin system and after all tests a virgin system will be restored. Could it be that something is wrong?

argilo, 2021-01-11:
It seems there's a problem in the gcc formula; 07722f21d0be2b63ae4e5d6f34601ece8625afe6 should work around the problem for now.