gqrx-sdr/gqrx #933: PlutoSDR support (macOS / pre-compiled)

chillophil, 2021-01-09:
I'm curious whether the precompiled macOS build supports the PlutoSDR, since I can't get it to work, but gnuradio works fine on my machine (all runtime dependencies are installed).

Maybe someone can give me some advice.

Thanks in advance!

argilo, 2021-01-11:
The release builds don't currently have Pluto support. I'll see about adding that.

chillophil, 2021-01-11:
Alright, in the meantime I discovered why I was not able to connect my PlutoSDR with Gqrx but with "vanilla" GnuRadio: The app package contains all the runtime files on its own - so it is not using the machine wide GnuRadio installation and plugins but the provided/packages ones.

Maybe I can add PlutoSDR support back again on my own and raise a pull request for that - if I find some free (spare) time.

As far as I know, PlutoSDR support was previously realized via gr-osmosdr-gqrx on branch "plutosdr" - is this still valid or does it need a complete "rework"?

I am asking, because I would like to add PlutoSDR support via gr-iio instead, so we can keep "vanilla" qr-osmosdr for all the other devices.

argilo, 2021-01-11:
> As far as I know, PlutoSDR support was previously realized via gr-osmosdr-gqrx on branch "plutosdr" - is this still valid or does it need a complete "rework"?

I think a better approach would be to ship the [SoapyPlutoSDR](https://github.com/pothosware/SoapyPlutoSDR) module with the Gqrx disk image. I tried that yesterday (https://github.com/csete/gqrx/compare/master...argilo:add-pluto) but for some reason dylibbundler was not bundling its libiio and libad9361 dependencies.

chillophil, 2021-01-12:
I agree with you, that seems to be a good, if not the best, solution!

I will dig into the dylibbundler issue if you haven't already - so let me know ;)

argilo, 2021-01-12:
Yes, I'd appreciate help with dylibbundler.

ulrichloose, 2021-01-12:
libiio / libad9361 are not dylibs and can not be bundled with dylibbundler...

libiio is a framework installed in library/frameworks via sudo

ulrichloose, 2021-01-13:
Another small problem: You would have to change SOAPY_SDR_ROOT in the info.plist so that the driver that comes with the bundle of gqrx is found. However, this means that the user is then off other soapydrivers that are probably in /usr/local/lib/SoapySDR/modules0.X

Whether this is still a good idea to pack the three Pluto files into the gqrx bundle would have to be questioned.

In principle, however, it works to redirect the paths of the soapypluto.so driver to the gqrx bundle with three lines install_name_tool. dylibbundler seems to be really only responding to the ending .dylib

chillophil, 2021-01-13:
As far as I am concerned, I don't care that much whether the libraries are bundled or not.

So generally speaking it would make a lot more sense to utilize "machine wide installed" libraries (if available) rather than the bundled ones and fallback to bundled resources if no local installation was found.

At least that was my first thought when I run Gqrx for the first time and I wondered why I could use my PlutoSDR with GnuRadio but not with Gqrx - since I realized that Gqrx ships its own environment.

maichai, 2021-07-01:
Are we getting anywhere with this soon? Although libiio, soapysdr and the SoapyPlutSDR module are globally installed and work fine, GQRX does not use them and thus does not detect my PlutoSDR.

chilippso, 2021-07-01:
@maichai Have you already tried the latest "master-branch" build ([Build #136](https://github.com/csete/gqrx/actions/runs/855306227)) ?

You can get the corresponding artifact (DMG) [here](https://github.com/csete/gqrx/suites/2774678160/artifacts/61496317) or pick it manually from the [Build Artifacts](https://github.com/csete/gqrx/actions/runs/855306227).

Unfortunately, I had not enough time for investigating further into this topic / issue.

maichai, 2021-07-02:
Thank you, I tried the first link, and yes it works, the Pluto shows up as "unknown".
Why not release that version officially?

chilippso, 2021-07-02:
@maichai I guess either it is not tested enough (yet) or there were some other (major / minor) changes that need to be reviewed / tested before releasing a new (official) version.

Nevertheless, this issue will be left open until there is an official released version with PlutoSDR support - even though it seems that we already have, but in "development state".

maichai, 2021-07-12:
actually I found the version you gave me to be non-functional with rtlsdr. Other tools work, either using the native interface, or using soapy.

argilo, 2022-01-28:
Gqrx versions 2.14.6 and later have working PlutoSDR support on MacOS (thanks to @ulrichloose's help in #937), so I'll close this issue off now.

closed 2022-01-28 by argilo

argilo, 2022-01-28:
I opened issue #1095 for the issue about the PlutoSDR showing up as "Unknown" in the device list.