gqrx-sdr/gqrx #938: AM-Stereo demod mode (C-quam)

kargaroc, 2021-01-18:
I think this would be nice to have!

AsciiWolf, 2022-02-27:
This would be great to have! I think it could be implemented as a part of the "AM-Sync" mode that is already present in Gqrx.

argilo, 2022-11-03:
C-QUAM doesn't look difficult to demodulate in software. However, C-QUAM stations are rare, and I don't have any in my area. If someone could make a good-quality recording of a C-QUAM station, that would be useful for testing.

mpdx-at, 2023-04-06:
Hey, I would also be interested to demodulate C-QUAM with gqrx.
What kind of record do you need - I can make a recording from an Italian AM Stereo station in the evening hours!

argilo, 2023-04-06:
I would need an I/Q recording (for example, made using Gqrx's built-in I/Q tool), and the signal to noise ratio would need to be high.