gqrx-sdr/gqrx #939: FUNcube Dongle V1.0 not working

reald, 2021-01-23:
If funcube dongle is configured in gqrx right after starting a message appears: "boost::too_many_args: format-string referred to fewer arguments then were passed - Please select another device". Funcube dongle will not work.

Funcube version: Funcube dongle pro 1.0 (the first one, NOT the Pro+)
Device string: fcd=0
gqrx Verision: 2.14.4.4 (07722f21d0be2b63ae4e5d6f34601ece8625afe6)
Ubuntu 20.04.1 LTS
Kernel: Linux blank 5.4.0-64-generic #72-Ubuntu SMP Fri Jan 15 10:27:54 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

dl9sau, 2021-05-01:
Same for me. My Funcube Dongle Pro does not work with current gqrx version  2.14.4 on macOS Big Sur v11.3.
"No supporded devices found (check the connection and/or udev rules).\nPlease select another device.".
"Configure I/O devices" opens. Choosabvle device stays on "FUNcube Dongle V1.0". Device string "fcd,type=1,device='FUNcube Dongle V1.0'".

Did work with "previous" version (https://gqrx.dk/news/gqrx-2-11-5-released).

DF3AK, 2023-07-04:
I have a similar problem with the same hardware.
Got rid of the boost::too_many_args: format-string..." error by installing osmocom-sdr v0.2.15.1 with gnuradio 3.10.5.1 from source. This requires the new FCD library gr-funcube, I've installed v3.10.0.rc2.

But now there is another issue. If I start osmocom_fft, everything works:
`Using FUNcube Dongle V1.0 (hw:1)
Funcube Pro :info: Start init fcd
Funcube Pro :info: Audio device hw:1 opened
fcd_control :info: FunCube Dongle  V1.0 initialized.
fcd_control :info: Dongle: FCDAPP 18.10 
`

The FFT-window opens and the spectrum looks decent.

If I start gqrx however, it seems like the audio source hw:1 is opened twice (once by osmocom/gr-funcube and a second  time by gqrx?). From the CLI

`Using FUNcube Dongle V1.0 (hw:1)
Funcube Pro :info: Start init fcd
Funcube Pro :info: Audio device hw:1 opened
fcd_control :info: FunCube Dongle  V1.0 initialized.
fcd_control :info: Dongle: FCDAPP 18.10 
fcd_control :info: LNA gain set to: 20
fcd_control :info: Mixer gain set to: 12
fcd_control :info: Set Frequency to: 140000000  Hz
fcd_control :info: LNA gain set to: 20
fcd_control :info: Mixer gain set to: 12
fcd_control :info: LNA gain set to: 20
fcd_control :info: Mixer gain set to: 12
fcd_control :info: Set Frequency to: 144482656 Hz
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file fcd rtl rtl_tcp rfspace redpitaya 
Using FUNcube Dongle V1.0 (hw:1)
Funcube Pro :info: Start init fcd
audio_alsa_source :error: [hw:1]: Device or resource busy
Funcube Pro :info: Could not open device: hw:1
audio_alsa_source :error: [hw:1,0]: Device or resource busy
`
Maybe gqrx is incompatible with newer versions of osmocom-sdr/gr-funcube?