gqrx-sdr/gqrx #940: gqrx.dk is out of date and doesn't have hint to use github instead

gdt, 2021-02-01:
I knew there have been recent releases and went to update the pkgsrc-wip package from 2.11.5 and went to gqrx.dk which indicates that 2.11.5 is still the latest release.  I eventually found this repo and recent releases.

I realize it's a lot of work to keep another site going, but a NEWS entry that says "For the latest release, see https://github.com/csete/gqrx/" would help a lot.

argilo, 2021-09-17:
Unfortunately I don't have the ability to make changes on gqrx.dk.

@csete What would you like to do with the site? I'd be happy to help maintain it if I had an account.

csete, 2021-09-26:
I have updated the download page and also disabled the Ubuntu and Raspberry Pi menus.

@argilo I'd be happy to give you access to the website. I will send you an email.

BerndGaykema, 2021-09-26:
Are you saying that GQRX latest versions (after 2.11.5) are no longer available for the Rapberry Pi?

argilo, 2021-09-26:
Gqrx 2.11.5 is available right in Raspberry Pi OS's package manager (`sudo apt-get install gqrx-sdr`), so there's no point downloading it from gqrx.dk.

argilo, 2021-09-26:
Subsequent versions have not been built for Raspberry Pi. If we make builds available again, we'll post them for download.

argilo, 2021-09-26:
Also, once Raspberry Pi OS updates to Debian 11, it will have the latest release of Gqrx (2.14.4).

argilo, 2021-10-04:
The site is now up to date (including a release notice for 2.14.5), so I'll close this off.

closed 2021-10-04 by argilo