gqrx-sdr/gqrx #941: Compiling problem

vnnspace, 2021-02-09:
Hello, I have a Ubuntu 20.04.
I am trying to compile GQRX on it already several days (I need help).
************************************************************************
if (d_decim >= 2 )
{
tb->disconnect(src,0,input_decim,0);
tb->disconnect(input_decim,0,iq_swap,0);
}
else
{
tb->disconnect(src,0,iqswap,0);
}
*************************************************************************
it write me for [ tb->disconnect(src,0,input_decim,0); ] no matching member function for call to 'disconnect'
same for [ tb->disconnect(input_decim,0,iq_swap,0); ] no matching member function for call to 'disconnect'

also exist the same problem for 'connect' function... 
Please guide me on what to do?

argilo, 2021-02-09:
Have you installed the `gnuradio-dev` package?

vnnspace, 2021-02-09:
Looks like it is installed
***************************************************************************
XXXX@GroundStation:~/osmo-bsc$ sudo apt install gnuradio-dev 
[sudo] password for ground: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gnuradio-dev is already the newest version (3.9.0.0git~master~14059~focal-4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
XXXX@GroundStation:~/osmo-bsc$

ulrichloose, 2021-02-09:
I don't have a Ubuntu, but on my Mac I can't compile gqrx with gnuradio in version 3.9. If I see it correctly, there have been quite far-reaching changes to gnuradio. With version 3.8.2 there are no problems - except the iq-balance in the head version now expects gnu-radio 3.9 as a minimum requirement and one should also choose a previous version.

vnnspace, 2021-02-09:
Does it mean you suggest to me change gnuradio to the 3.8.2? and try?

ulrichloose, 2021-02-09:
In the meantime i got it on my mac running with 3.9... You have to update libgnuradio-osmosdr to the newest version too...

```
/usr/local/include/gnuradio/hier_block2.h:137:5: note: candidate function not viable: no known conversion from
      'osmosdr::source::sptr' (aka 'shared_ptr<osmosdr::source>') to 'gr::basic_block_sptr' (aka 'shared_ptr<gr::basic_block>') for
      1st argument
    disconnect(basic_block_sptr src, int src_port, basic_block_sptr dst, int dst_port);
```

vnnspace, 2021-02-09:
It is the final version 
++++++++++++++++++++++++++++++++++++++++++++++++++++++++
sudo] password for ground: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgnuradio-osmosdr0.2.0 is already the newest version (0.2.0-2).
libgnuradio-osmosdr0.2.0 set to manually installed.
The following packages were automatically installed and are no longer required:
  freeglut3 libcppunit-1.15-0 libcppunit-dev libglfw3 libgnuradio-analog libgnuradio-audio libgnuradio-blocks
  libgnuradio-channels libgnuradio-digital libgnuradio-dtv libgnuradio-fec libgnuradio-fft libgnuradio-fft3.8.1
  libgnuradio-filter libgnuradio-fosphor3.8.0 libgnuradio-network libgnuradio-pmt libgnuradio-qtgui
  libgnuradio-runtime libgnuradio-trellis libgnuradio-uhd libgnuradio-video-sdl libgnuradio-vocoder
  libgnuradio-wavelet libgnuradio-zeromq libgsm1-dev libjs-jquery-ui liblbfgsb0 python-matplotlib-data
  python3-click-plugins python3-cycler python3-decorator python3-kiwisolver python3-matplotlib python3-networkx
  python3-opengl python3-pyqt5.qtopengl python3-pyqtgraph python3-scipy python3-zmq uhd-host
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

ulrichloose, 2021-02-09:
v0.2.0 seems not the newest one... There is a 0.2.2 and a 0.2.3 but all three are for GR-3.8... Only the Master is for GR-3.9.

So there are only two ways - use not GR-3.9 or try to get libgnuradio-osmosdr made from the Master.

argilo, 2021-02-09:
Do you care about having the very latest version of Gqrx? If not, just install the `gqrx-sdr` package and use that.

vnnspace, 2021-02-11:
Thanks, everyone to try to support me. The main issue that is, I am using BladeRF with LNA and PA connected directly to the SDR, and I can not understand how to turn on bias by using GQRX? I thought maybe in a new version this feature will be realised... If not, it is no meaning for me to use it.
If I am using BladeRF-cli ( here not a problem to turn on BIAS )
Maybe you can advise how I can turn ON bias on BladeRF by using the software? Maybe I should put some FLAG to 1 in some file before starting to compile GQRX? But if I will install GQRX by using apt ( I can not do any compiling ) and I am already installing it before and I am not found how to turn on BIAS on it ...

argilo, 2021-02-11:
It looks like you should be able to turn on the bias tee by using the following device string:

`bladerf=0,biastee=1`

vnnspace, 2021-02-11:
Thank you but it is not working like this ( or am doing something wrong ) 
I have picture 
[image]
and line in Device string is 
bladerf=0,buffers=128,buflen=8192,transfers=32
If I am writing 
bladerf=0,buffers=128,buflen=8192,transfers=32,biastee=1 (nothing change )
*****************************************
[bladeRF common] init: DEBUG: entering initialization
[bladeRF source] init: Opening Nuand bladeRF with device identifier string '*:instance=0'
[bladeRF source] Device: Nuand bladeRF 2.0 Serial # 937b...1e56 FW v2.4.0 FPGA v0.12.0
[bladeRF source] init: Buffers: 128, samples per buffer: 8192, active transfers: 32
[bladeRF source] bladerf_source_c: DEBUG: initialization complete
**********************************************************************************
and after rebooting GQRX biastee =1 sign desapeare from device string......
where I can read what I can write in device string ?

argilo, 2021-02-11:
Device string options are listed here:

https://osmocom.org/projects/gr-osmosdr/wiki#bladeRF-Source-Sink

It seems the `biastee` option is not documented there, but it was added to gr-osmosdr in version 0.1.5:

https://github.com/osmocom/gr-osmosdr/commit/c06db96489f8a314a859afb9e96cb3ffcc62ad3a

vnnspace, 2021-02-11:
If i am write biastee=1 or biastee=0 or biasteerx1=1 or biasteerx1=0 nothing change at all....
Maybe I do something wrong?

argilo, 2021-02-11:
`biastee=1` should be the correct syntax. Unfortunately I do not have a bias tee to test with, but I do see an error message about not having a bias tee when I enable the option.

vnnspace, 2021-02-11:
Thank you. You absolutely right when I write biastee=1 I do also not see any error messages but changes also not see. But when I disconnect LNA or PA my signal became normal, and when I use BladeRf-cli also everything became normal.... that is why it looks very very strange

vnnspace, 2021-02-16:
Thank you again to everyone. I investigated that if I write biastee=1 only RX bias is turned ONTX bias is not, which is bad. I will try to change CODE in gqrx to add this opportunity to it.

closed 2021-02-16 by vnnspace

argilo, 2021-02-16:
It is very likely that the issue lies in gr-osmosdr, not Gqrx.