gqrx-sdr/gqrx #942: Feature request: Dynamic bookmarks for Shortware stations.

stdevPavelmc, 2021-02-15:
When you are spinning the dial in HF bands (3-30MHz) there are a lot of broadcasting stations with shifting frequencies and schedules most of the time the user has to go to an online service to identify the station, set a bookmark for it... to find that half an out later there is another station on the same frequency, very annoying.

What if gqrx implement a feature called "**SWL auto bookmarking**"? I would love it and a lot of SWL out there two.

There are a few updated list on the internet for schedules, languages, target areas, etc; for example this one: http://www1.s2.starcat.ne.jp/ndxc/ and direct txt file http://www1.m2.mediacat.ne.jp/binews/us/userlist1.txt; that lists are updated at least once a month, some times weekly.

The mentioned feature will be triggered when a user is seeing the HF spectrum (3-30 Mhz, display frequency as we can use a up-converter) and has indicated that the want it on (button?)

Up on activation gqrx can fetch the list (or use a earlier cached version, less than a week old) to parse the actual date and time to create a custom bookmark file that will be loaded internally and shown on the spectrum view.

For example a bookmark can contain this info: `WRMI (Viva Miami, Spa, 0200-0300z)` in reference to WRMI Station, Broadcasting "Viva Miami" on Spanish language from 0200 to 0300 UTC time.  (time can be configurable to show local time instead of UTC)

That will give the user the expected broadcast on the display, that will change as one broadcast station shutoff and he other came alive on the same frequency (very common).

I can visualize another uses for a feature like that on a next evolution, for example loading repeaters information on 29Mhz, 50Mhz, 145Mhz, 440Mhz and others for ham radio people from online services, of curse you will need to specify your location via a grid locator or plain coordinates.

Now thinking on the future, in next iterations gqrx can offer the possibility to the user to shown and reserve the recording of programs based on that list, just dreaming.

Cheers, Pavel CO7WT

shuyuan-liu, 2021-02-16:
That sounds like a nice feature but I've got a difeuferent idea on how to implement it.

Write an separate program that queries the current frequency through the remote control feature, changes bookmarks.xml based on time, and make gqrx able to reload bookmarks whenever bookmarks.xml changes.

I think this is a more "modular" approach because dynamic bookmarks is a feature useful only to *some* users of GQRX: others might not need them and having it in GQRX just increases compile time and file sizes.

What do you think?

stdevPavelmc, 2021-02-16:
I Agree, It can be modular.

I can program the companion software in python and offer binaries for different arches, but I will need a few hints:
- Can GQRX actually reload the bookmarks.xml up on file modification? (It works now or will requite a issue to implement that?)
- Where I can get details (docs) of the remote control interface to query center freq and span of the waterfall

Cheers.

shuyuan-liu, 2021-02-16:
> Can GQRX actually reload the bookmarks.xml up on file modification? (It works now or will requite a issue to implement that?)

I tried and it looks like gqrx doesn't reload by itself. I haven't looked into the code, but [QFileSystemWatcher](https://doc.qt.io/qt-5/qfilesystemwatcher.html) might be useful if it needs to be implemented. I searched the repo for this class and it seems it's not used anywhere.

> Where I can get details (docs) of the remote control interface to query center freq and span of the waterfall

The page [Controlling gqrx from a remote host – Gqrx SDR](https://gqrx.dk/doc/remote-control) has a slightly outdated list of commands. Everything listed on there still works, but it is missing new commands like set and get VFO. You can always look in [the code handling remote control commands](https://github.com/csete/gqrx/blob/master/src/applications/gqrx/remote_control.cpp) where all the commands and corresponding functions are listed. I don't think there is a command to get current freq span of the waterfall but it shouldn't be too difficult to code one in.

stdevPavelmc, 2021-02-17:
Ok, thanks for the tips

I'm readding and codding, I'm working to parss the file http://www1.m2.mediacat.ne.jp/binews/us/userlist1.txt and creating the query sentences to  get the active only station on the current time.

For now the only needed feature of gqrx is the auto loading of the bookmarks file if this changes, the checks can be every minute after the 45 second mark.

In that way we can see the changes in the waterfall seconds after it changes, and so, broadcasting stations uses to tune in the tx sites for about 30 seconds before transmissions.

stdevPavelmc, 2021-02-17:
It's working no, see attached images..

I need to close & open GQRX to be able to update the bookmarks

Will file an issue with this specific request: Reload the bookmarks.csv file it it changes (1 minute timer)

[image]

[image]

[image]