gqrx-sdr/gqrx #946: Feature: Multiple demodulators simultaneously

kord1e, 2021-03-02:
The ability to set up multiple demodulators simultaneously might be somewhat useful, for example, when one wants to listen to several air traffic channels at once.

KevWal, 2021-03-25:
A plus one from me, the ability to run 2 copies of a decoder like wsjtx, linked to two different audio streams out of Gqrx, to decode FT8 and WSPR for example for my balloon tracking.

LouDou, 2021-09-21:
I've started looking into this on a branch here: https://github.com/LouDou/gqrx/tree/multirx

It took about 4hrs work to de-tangle and re-work the "receiver" controller class, and I've proven that the DSP graph is capable of doing this - with each receiver outputting to its own audio stream (when using pulseaudio or portaudio).

None of the GUI work has been done yet, and I've had to disable all recording, RDS, AFSK features too - in fact, I don't even know the best way to represent the multiples of all the receiver settings... it's fundamentally quite intrusive changes with regards to using the application. Also, there's quite a bit of work to do around adding/removing receivers without completely breaking everything.

... but it's a start.

LouDou, 2021-09-22:
[image]

Well, lets demodulate 2 broadcast FM stations at once then :)

Still lots to do, but some basic control over two Receivers is possible here.