gqrx-sdr/gqrx #947: Feature: Selection menu for RTL-SDR options such as direct sampling

kord1e, 2021-03-03:
It would be useful to have drop-down menus or something similar to select different options for RTL-SDR, for example, direct sampling. At least for the most commonly used options. Now one must write these options into the text box.

ulrichloose, 2021-03-05:
If it is only a matter of switching directsampling, this is possible without a change in gqrx. The librtlsdr from hayguen is far more advanced as the original and contains an automatic that switches independently when exceeding / falling below a frequency.

https://github.com/ulrichloose/librtlsdr with a small bug fix.

To do this, the environment variable LIBRTLSDR_OPT must be set to dm=4. direct_samp=2 must be entered as a gqrx device-string. After that, the rtl-stick automatically switches below 30 MHz to directsampling and vice versa. I've been using this for a long time and it works without any problems - as if you had a continuous range of 0 - well over 2.4 GHz.

DG2YCB, 2021-06-10:
I am trying to bring Gqrx to work on HF. I have installed the librtlsdr from ulrichloose and set my RTL-SDR device string to "direct_samp=2". However, I cannot tune below 24 MHz. 
What else is needed?
What exactly do I have to do to set " the environment variable LIBRTLSDR_OPT to dm=4?
And further question: How can I modify it so that automatically switching is done at 24 MHz instead of 30 MHz? (I like to do RX on 12m and 10m band with direct sampling disabled).
You can also send me your answer by email. (Email address on qrz.com)

ulrichloose, 2021-06-11:
In addition to the device string, "No Limits" must be activated in the input controls of gqrx, otherwise gqrx refuses to skip the direct sampling limit. With the environment variable LIBRTLSDR_OPT, options can be passed to librtlsdr that are otherwise not supported by e.g. gqrx.

The switching threshold is stored in the driver depending on the tuner model - with my R820T tuner this is 24 MHz. However, it should be possible to transfer this value with dm = as well:

dm = <ds_mode_thresh> set dynamic direct threshold mode or threshold frequency:

0: use I & Q
1: use I
2: use Q
3: use I below threshold frequency
4: use Q below threshold frequency (= RTL-SDR v3)
other values ​​set the threshold frequency

As an example: export LIBRTLSDR_OPT=dm=4:dm=30000000, here it is switched at 30 MHz. When librtlsdr reads this, the following is displayed on the command line of gqrx:

process options 'dm = 4: dm = 30000000' from environment 'LIBRTLSDR_OPT'

rtlsdr_set_opt_string (): parsed direct sampling mode 4 == 4: use Q below threshold frequency
Disabled direct sampling mode
  application of parsed option returned 0

rtlsdr_set_opt_string (): parsed direct sampling threshold 30000000
Enabled direct sampling mode, input 2
  application of parsed option returned 0
Allocating 15 (non-zero-copy) user-space buffers

#### here it is switched back and forth between frequencies below 30 MHz and above 30 MHz ####

Disabled direct sampling mode
Enabled direct sampling mode, input 2
Disabled direct sampling mode
Enabled direct sampling mode, input 2
Disabled direct sampling mode
Enabled direct sampling mode, input 2
Disabled direct sampling mode
Enabled direct sampling mode, input 2

The whole thing works fine for me - but the original librtlsdr from hayguen has a small bug that I fixed with my fork. I would use the "localtest" branch. It is content with a 32-bit frequency setting - the latest version of hayguen uses 64-bit and that is implemented a bit shaky from my point of view.

DG2YCB, 2021-06-11:
Many thanks. Very helpful! I followed exactly your description.

GQRX as well as the RTL-SDR device are working well meanwhile. However, automatic switching does not happen. Why? No other librtlsdr driver is installed than the one compiled from https://github.com/ulrichloose/librtlsdr. 

I can manually switch to HF when in GQRX I set the device string to either rtl=0,direct_samp=2 or rtl=0,direct_samp=3, and back to VHF with rtl=0,direct_samp=0 or just rtl=0. "No Limits" enables to tune to any frequency, but as said my RTL-SDR device is NOT AUTOMATICALLY SWITCHING. 

Also the variable seems to be set correctly, because with echo $LIBRTLSDR_OPT I get dm=4:dm=24500000 (that's what I have set).

What may be the reason why auto-switching does not work?

And maybe I am running into another issue: When I manually switch between HF and VHF, GQRX each time crashes with the following error message: "gqrx exited with an exception: Failed to open rtlsdr device."

Any idea?

ulrichloose, 2021-06-11:
Hard to say why this doesn't work ...

You will have to feel your way around ... What certainly cannot work is the use of a finished app image - gqrx has to be compiled locally so that the correct librtlsdr is used. That is why it is not enough to rely on an echo ... when troubleshooting. The locally compiled gqrx must be called from the commandline so that you can get the information from the librtlsdr:

process options 'dm = 4: dm = 30000000' from environment 'LIBRTLSDR_OPT'

rtlsdr_set_opt_string (): parsed direct sampling mode 4 == 4: use Q below threshold frequency
Disabled direct sampling mode
application of parsed option returned 0

rtlsdr_set_opt_string (): parsed direct sampling threshold 30000000
Enabled direct sampling mode, input 2
application of parsed option returned 0

would be the feedback that should be displayed by librtlsdr. If this is not displayed, a wrong librtlsdr is being used and then switching cannot work.

In terms of "gqrx exited with an exception: Failed to open rtlsdr device." unfortunately I have to fit - I'm using gqrx on a Mac and I haven't seen this error message yet. Maybe another user can help?

However, it would certainly make sense to clarify this behavior in advance:

direct_samp = (0 = no direct sampling; 1 = direct sampling on I, 2 = direct sampling on Q.)

With manual entry in gqrx, rtl=0,direct_samp=0 or rtl=0,direct_samp=2 would normally be correct and this must not lead to a crash.