gqrx-sdr/gqrx #953: Main frequency control widget doesn't work properly

dvdesolve, 2021-04-04:
At first I want to post [a link](https://www.youtube.com/watch?v=MECuNU2RGgE) to the video I've captured so my further explanation will be clearer.

I haven't been using GQRX for about a half of an year and now I've faced some strange bug. I'm using latest Arch Linux and GQRX version is 2.14.4. For the example video I've erased the whole config directory just to start from a scratch. As you can see on a video I've selected my RTL SDR dongle and set sampling rate to 2.4 Msps without decimation (but the same behavior is observed for other rates and decimations too). After that I've set main frequency to 91.2 MHz (local FM radio broadcasting station), gain to approx 12 dB and set DC remove feature. However, as you can see (0:35) after starting DSP processing the receiver was on incorrect frequency (the first appearance of a bug). Some mangling with frequency control allowed me to go for a desired frequency range. And starting from 0:57 you can see the bug in full action: if I set main frequency quite slowly with my mouse there are no changes on waterfall, FFT and in reception despite numbers on FFT axis are changing. But if I change frequency by fast mouse scrolling we can observe its change. However, even in this case we're not ending on the right frequency. It's even more interesting (1:43) that supporting frequency control widget on a right of a screen works as intended. Also dragging the bandwidth with a mouth works too.

Regarding the above facts I think that there is a bug in a main frequency control widget.

dvdesolve, 2021-04-18:
Last working version (IIRC) was 2.12.1. Just checked with 2.13 - the bug is here

dvdesolve, 2021-04-18:
Okay, seems like it's related to the RTLSDR driver somehow. When run from command line I observe the following:
```
[19:17 viktor@desolve-desktop ~]$ gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
BandPlanFile is /home/viktor/.config/gqrx/bandplan.csv
BookmarksFile is /home/viktor/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 10.2.0; Boost_107500; UHD_4.0.0.0-0-unknown
Found Rafael Micro R820T/2 tuner
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T/2 tuner
ATTENTION: default value of option allow_rgb10_configs overridden by environment.
Allocating 15 zero-copy buffers
rtlsdr_demod_write_reg failed with -9
r82xx_write_arr: i2c wr failed=-9 reg=17 len=1
rtlsdr_demod_write_reg failed with -9
r82xx_write_arr: i2c wr failed=-9 reg=17 len=1
rtlsdr_demod_write_reg failed with -9
r82xx_write_arr: i2c wr failed=-9 reg=17 len=1
...
```

Lines with errors appear on scrolling events

closed 2021-06-10 by dvdesolve

cybekRT, 2025-01-31:
Even if this issue is related to the USB and other library, shouldn't UI be aware of the result of function and do not change displayed values if the frequency wasn't properly changed?