gqrx-sdr/gqrx #954: MacOS: not compatible with case-sensitive file system

pavelrevak, 2021-04-07:
On my mac I have only case-sensitive FS and Gqrx does not start - with any message
If Gqrx is started from terminal, there is error, that an image not found...
```
$ Gqrx.app/Contents/MacOS/gqrx 
dyld: Library not loaded: @executable_path/../libs/libboost_date_time-mt.dylib
  Referenced from: /Applications/Gqrx.app/Contents/MacOS/gqrx
  Reason: image not found
zsh: abort      Gqrx.app/Contents/MacOS/gqrx
```

in `Gqrx.app` is directory `Libs` and not `libs`, after rename, the app started and seems is working

ulrichloose, 2021-04-07:
Oh you're the Mac user who uses a case-sensitive FS  (:-)

This is causing problems from Photoshop to Gqrx... However, this can certainly be easily corrected in macos_bundle.sh:

`dylibbundler -s /usr/local/opt/icu4c/lib/ -od -b -x Gqrx.app/Contents/MacOS/gqrx -x Gqrx.app/Contents/soapy-modules/libremoteSupport.so -d Gqrx.app/Contents/libs/`

argilo, 2021-04-08:
Fixed in 6fce5aae8a906b6b7e067b0e1515c5d23fe3b0ae.

Thanks for the bug report @pavelrevak and for the suggested fix @ulrichloose!

argilo, 2021-04-08:
@pavelrevak: You can find an updated build here: https://github.com/csete/gqrx/actions/runs/727842389. I tested this by copying it into a case-sensitive disk image and running it from there, but I'd appreciate if you could confirm that it works for you as well.

closed 2021-04-08 by argilo

pavelrevak, 2021-04-08:
Thanks @ulrichloose and @argilo for very quick fixup, now everything is working! also thanks for MacOS builds.