gqrx-sdr/gqrx #959: GQRX freezes with hackRF 2021-0

Potomac, 2021-04-24:


Potomac, 2021-04-25:
If I use the official appimage of Gqrx (last version, same as archlinux package : 2.14.4) then there is no freeze, no bug with hackRF 2021-03-1 driver :
https://github.com/csete/gqrx/releases

Potomac, 2021-04-25:
But the appimage can also contain hackRF lib in 2018-01 version, that can explain why there is no freeze.

Potomac, 2021-04-28:
I made a git bisect and I found the faulty commit in the hackrf lib : 

```
1442014a80b5ce33f16ef665538c2881b37d3a68 is the first bad commit
commit 1442014a80b5ce33f16ef665538c2881b37d3a68
Author: Matioupi <mathieu.peyrega@gmail.com>
Date:   Wed Oct 16 12:03:08 2019 +0200

    Modified hackrf_stop_tx and hackrf_stop_rx to first join the transfer thread
    before setting the receiver to OFF mode (cf. issue #650)

 host/libhackrf/src/hackrf.c | 9 +++++----
 1 file changed, 5 insertions(+), 4 deletions(-)
```
https://github.com/mossmann/hackrf/commit/1442014a80b5ce33f16ef665538c2881b37d3a68

Potomac, 2021-04-29:
Sadly reverting this commit from hackrf is not enough for solving freezes in Gqrx, as recent commits in hackrf change also start/stop rx, tx functions,
1442014 is just the first bad commit, the problem may be more deep, related to thread management, usb transfert, it will need more debugging for solving this bug.

Potomac, 2022-02-04:
The bug is fixed by these commits made by Martin : 
https://github.com/greatscottgadgets/hackrf/pull/1029

The bug was due to a race condition triggered by 2021.3 driver and firmware version of hackRF.

argilo, 2023-04-30:
Closing, as the issue appears to have been resolved upstream.

closed 2023-04-30 by argilo