gqrx-sdr/gqrx #961: Fractional PPM setting seems broken?

devnulling, 2021-06-04:
Support for non int PPMs was added here: https://github.com/csete/gqrx/pull/201

However, when trying to set 0.1 or 0.x PPM values, there appears to be no change. Tested with a USRP and RTLSDR.

Tested on master 272bd75e8cb33ff54d2d8f71295295dcdbf4b962

argilo, 2021-10-04:
I tested with a USRP, and fractional PPM values worked fine. Could you try again and let me know if it works for you?

For RTL-SDR, fractional PPM will not work because librtlsdr accepts the PPM value as an `int`: https://github.com/steve-m/librtlsdr/blob/d770add42e87a40e59a0185521373f516778384b/include/rtl-sdr.h#L162

argilo, 2021-10-12:
I'll close this for now, since fractional PPM appears to be working. @devnulling Feel free to reopen if you fin otherwise.

closed 2021-10-12 by argilo

helioshk, 2023-11-07:
It's not working for me. -1.0 ppm is the same correction as -1.9 ppm. +1.0 ppm is the same as +1.9 ppm

Gqrx 2.15.9 from debian with RTL-SDR blog v4 dongle.
And while we're at it, 0.01 ppm steps would be nice too ;)

argilo, 2023-11-07:
As mentioned above, the RTL-SDR driver does not support fractional PPM, so Gqrx's hands are tied. Someone should contribute a fix to the RTL-SDR driver and gr-osmosdr.