gqrx-sdr/gqrx #963: GNU Radio UDP sink is now under gr::network::udp_sink in the gnuradio-network module

fventuri, 2021-06-26:
Hello,
I am running GNU Radio from the master branch here (version 3.10), and I noticed that gqrx does not build anymore because they moved the UDP sink block under `gr::network::udp_sink` (it used to be `gr::blocks:udp_sink`). They also added the GNU radio core module `gnuradio-network` where the UDP sink functions like `gr::network::udp_sink::make()` are defined.

I made a few changes to the gqrx source code and cmake config files to build it with the current GNU Radio master branch (`v3.10.0.0git-429-g655e232a`); after  built it, I ran a quick test with gqrx here a few minutes ago, and I was able to receive a local FM station.

I just pushed my changes here: https://github.com/fventuri/gqrx/tree/network_udp_sink

If you are interested I can open a pull request to make gqrx build with the current GNU Radio master branch (version 3.10).

Regards,
Franco Venturi - K4VZ

phaethon, 2021-08-25:
Thanks @fventuri ! I needed to use Master branch of GNU Radio because of PlutoSDR recently brought into it. With your patch gqrx compiles successfully.

argilo, 2021-09-29:
I actually started some work on this already; see #958 for details.

I ran into a problem though: The UDP Sink block crashes if you stop & restart it: https://github.com/gnuradio/gnuradio/issues/5112

I think GNU Radio will need to fix that bug before Gqrx can start using the new block.

argilo, 2021-09-29:
Because the buggy UDP sink will be present in some released versions of GNU Radio (3.9.0 to 3.9.2), I should probably update my branch to switch to gr-network in 3.10 instead of 3.9.

argilo, 2021-09-29:
@ulrichloose You may be interested in this topic, since you tested out my branch a while back (https://github.com/argilo/gqrx/commit/1b19970f6fdf8ea30fa82a118c938a3a3cb003e7#commitcomment-50212951). I did a similar test just now and didn't have any trouble decoding pager traffic with multimon-ng. But I'll try making an I/Q recording and replaying it to see if there is any difference in the UDP packets between the old & new code.

argilo, 2021-09-29:
> But I'll try making an I/Q recording and replaying it to see if there is any difference in the UDP packets between the old & new code.

I tried this just now, and the outputs were identical, apart from the occasional rounding error in the least significant bit of a sample.

closed 2021-09-30 by argilo

fventuri, 2021-10-01:
@argilo - Clayton, first of all thanks for fixing this issue.

I built the latest GNU Radio from the master branch, then I built the 'gr-osmocom' module that supports SDRplay API version 3.X (https://github.com/fventuri/gr-osmosdr), and finally 'gqrx' from the repository master branch.
Everything compiled without errors, and I was able to run a quick test with gqrx to listen to a local FM BC station successfully.

Just a minor nitpick: I noticed that the README in this gqrx repository still refers to Alexandru Csete repository in the build instructions (and perhaps in other places); I was thinking that maybe they should be changed from 'csete' to 'gqrx-sdr'.

Franco

argilo, 2021-10-01:
Thanks for pointing that out. I've opened https://github.com/gqrx-sdr/gqrx/pull/983 to update the URLs.

PR #982 (2021-09-30): Use gr-network's UDP Sink in GNU Radio >= 3.10
Fixes #963.
Replaces #958.

Even though the new UDP Sink block is available in GNU Radio 3.9, I've limited this change to GNU Radio 3.10 to prevent https://github.com/gnuradio/gnuradio/issues/5112 from affecting GNU Radio 3.9 users. Until that bug is fixed, GNU Radio 3.10 users will experience a crash if Gqrx's DSP settings are changed during UDP streaming.

I tested this change against GNU Radio 3.8, 3.9 and 3.10. I used multimon-ng for testing.

/cc @fventuri @phaethon @ulrichloose @willcode