gqrx-sdr/gqrx #964: Long audio recording make .wav corrupted

aerogus, 2021-06-30:
I did a long audio recording (about 10 hours) with gqrx. The recorded .wav was big (7 GB) and I was unable to read it correctly (only the first 4 hours were playable). A .wav file can't be technically more than 4GB because of a 32 bits counter in the header which store the file length.

I managed to repair my recording splitting the big file in smaller parts, removing / regenerating the riff wave header.
I did an article (in french) explaining that story : https://aerogus.net/posts/reparer-fichier-wav-trop-long/

My suggestions:
- is it possible to limit the length in time (or the weight in bytes) of a recorded file ?
- is it possible to change the number of recorded tracks (mono instead of stereo) or the sample rate (32k instead of 48k) ?

drmpeg, 2021-07-02:
If you're using Gqrx built against GNU Radio 3.9, there is support for 64-bit WAV files.

If you're able to build from source, change this line:

https://github.com/csete/gqrx/blob/master/src/applications/gqrx/receiver.cpp#L1009

from:

gr::blocks::FORMAT_WAV, gr::blocks::FORMAT_PCM_16);

to:

gr::blocks::FORMAT_RF64, gr::blocks::FORMAT_PCM_16);

Of course, this should be made a configuration option.

ljkenny, 2021-10-07:
Just bumped into this myself.

Microsoft PCM WAVE files (the default) only appear to support 4GB since the 32bit data chunk gets set to 0 over that.

This means processing with `sox` is impossible.  All of my attempts result in a 44 Byte (header only - nil data) file.

Why not either make 64-bit support the default, or switch to something like *.mp3?

drmpeg, 2021-10-07:
Gqrx is a GNU Radio application, so it's always linked to GNU Radio. I was just suggesting that when someone gets around to adding recording format selection to Gqrx, that along with WAV64, FLAC and Ogg Vorbis could also be included.

But even if this change was made tomorrow, it won't show up in Linux distro repositories for quite a while. Even Debian Bullseye is still GNU Radio 3.8.