gqrx-sdr/gqrx #971: Compile error

HarleyWaagmeester, 2021-08-04:
Using Debian 11.
Wayland X.
When I attempt to compile gqrx (   cd build;  cmake ..;  make;   )
i get the following error from the gnu compiler:

gqrx/src/interfaces/udp_sink_f.cpp:100:36: error: no matching function for call to ‘udp_sink_f::connect(gr::blocks::float_to_short::sptr&, int, gr::blocks::udp_sink::sptr&, int)’
  100 |         connect(d_f2s, 0, d_sink, 0);

I don't see anything overtly incorrect in the code.
But i have sub-zero C++ skills.

And also from the gqrx-sdr debian 11 package, when i run gqrx i get this:
gqrx exited with an exception:
std::bad_alloc
and the console says 'Segmentation fault.'
Can anyone advise me about this ?
TNX

markjfine, 2021-08-04:
Strange that it would fail on the second `connect(d_f2s, 0, d_sink, 0);` and not the one in the `if (stereo)` block on line 95.

Anyway, all of this is supposed to interface to gnuradio, which I assume you have installed, otherwise it would barf on the includes in the .h file. Are you sure you are building with the correct version of gnuradio (>=3.8)?

argilo, 2021-09-17:
I'm not sure what's going on here. I installed Debian 11 in a virtual machine, and I was able to build gqrx from source without issue. I also didn't experience a segfault when using the pre-build gqrx-sdr package. Are there any other details of your setup that might be relevant?

The compilation error looks similar to the one reported in #952.

argilo, 2021-09-17:
A couple questions:
* Did `cmake ..` complete without errors?
* Do you have the gnuradio-dev package installed?

argilo, 2021-10-04:
Sine I haven't heard anything back, I'll close this off for now.