gqrx-sdr/gqrx #972: After installing GPRX, NESDR Mini 2 is no longer recognized on the USB port

SebastianStorb, 2021-08-08:
I tried to receive flight radar on a new Rasperi Pi. It worked very well on Windows so I wanted to bring it to a smaller device. At first I tried the following instructions without success:

https://www.instructables.com/FlightRadar-on-a-Raspberry-Pi/
https://osmocom.org/projects/rtl-sdr/wiki/Rtl-sdr

Then I wanted to test the receiver under Linux and came across gprx. After installing via "Add / remove Software", the software crashed after changing settings and the hardware was not recognized on the USB port under Windows or Linux.

Is there a simple or understandable solution to bring the device back to its old state?

argilo, 2021-10-04:
Were you able to get the device working again? I suspect the problem was unrelated to Gqrx.

SebastianStorb, 2021-10-04:
No - the device is unusable since this. It directly was broken after I used the software. 

Regards

argilo, 2021-10-04:
OK. I suspect it was a hardware failure then. Gqrx uses librtlsdr under the hood to interface with RTL-SDR devices, and that library is stable and well tested.

I don't see anything that Gqrx can do about this, so I'll close off the issue for now.

closed 2021-10-04 by argilo