gqrx-sdr/gqrx #973: RDS data character encoding?

cyberpunkrocker-zero, 2021-08-08:
What chararacter encoding Gqrx is using for RDS data?
As a Finland-living person, I see all special Scandinavian characters (å, ä, ö etc.) as an 'unknown character' in the RDS data. Most local stations have simplified things by replacing ä with a and so on in their RDS data, though, so I'm not sure whether this is an error from the specific station's part or not.

Arch Linux
gqrx-2.14.4
fi_FI.UTF-8

argilo, 2021-09-17:
The RDS code in Gqrx was copied over from [gr-rds](https://github.com/bastibl/gr-rds). I had a peek, and it looks like the code doesn't make any attempt to handle non-ASCII characters. It looks like RDS has its own character table, which would need to be added to gr-rds. If you could make an I/Q recording of one of your local stations displaying unknown characters, that would be very helpful for testing.

argilo, 2021-09-18:
It looks like the correct encoding is the one defined in Annex C (page 42) here:

https://www.etsi.org/deliver/etsi_ts/101700_101799/101756/01.06.01_60/ts_101756v010601p.pdf

A first crack at representing that in C++:
```c++
    const char *ebu[] = {
        "", "", "", "", "", "", "", "", "", "", "", "", "", "", "", "",
        "", "", "", "", "", "", "", "", "", "", "", "", "", "", "", "",
        " ", "!", "\"", "#", "¤", "%", "&", "'", "(", ")", "*", "+", ",", "-", ".", "/",
        "0", "1", "2", "3", "4", "5", "6", "7", "8", "9", ":", ";", "<", "=", ">", "?",
        "@", "A", "B", "C", "D", "E", "F", "G", "H", "I", "J", "K", "L", "M", "N", "O",
        "P", "Q", "R", "S", "T", "U", "V", "W", "X", "Y", "Z", "[", "\\", "]", "―", "_",
        "║", "a", "b", "c", "d", "e", "f", "g", "h", "i", "j", "k", "l", "m", "n", "o",
        "p", "q", "r", "s", "t", "u", "v", "w", "x", "y", "z", "{", "|", "}", "¯", "",
        "á", "à", "é", "è", "í", "ì", "ó", "ò", "ú", "ù", "Ñ", "Ç", "Ş", "ß", "¡", "Ĳ",
        "â", "ä", "ê", "ë", "î", "ï", "ô", "ö", "û", "ü", "ñ", "ç", "ş", "ğ", "ı", "ĳ",
        "ª", "α", "©", "‰", "Ğ", "ě", "ň", "ő", "π", "€", "£", "$", "←", "↑", "→", "↓",
        "º", "¹", "²", "³", "±", "İ", "ń", "ű", "µ", "¿", "÷", "°", "¼", "½", "¾", "§",
        "Á", "À", "É", "È", "Í", "Ì", "Ó", "Ò", "Ú", "Ù", "Ř", "Č", "Š", "Ž", "Ð", "Ŀ",
        "Â", "Ä", "Ê", "Ë", "Î", "Ï", "Ô", "Ö", "Û", "Ü", "ř", "č", "š", "ž", "đ", "ŀ",
        "Ã", "Å", "Æ", "Œ", "ŷ", "Ý", "Õ", "Ø", "Þ", "Ŋ", "Ŕ", "Ć", "Ś", "Ź", "Ŧ", "ð",
        "ã", "å", "æ", "œ", "ŵ", "ý", "õ", "ø", "þ", "ŋ", "ŕ", "ć", "ś", "ź", "ŧ", ""
    };
```

argilo, 2021-09-19:
The situation with character encodings seems to be a mess:

https://pira.cz/rds/rdschars.pdf

I tested the two physical receivers I have on hand, and neither of them uses the RDS code page; one uses ASCII and does not display characters with the high bit set, and the other uses the Windows-1252 code page! Also, one of my local radio stations is using ASCII rather than the RDS code page; if I switch gr-rds to use the RDS code page then a ` character incorrectly displays as ║.

It seems like there is not much hope of displaying things correctly without making the code page user-selectable.

argilo, 2021-09-19:
By the way, I see someone once tried to make this a user-selectable option in Gqrx: #565

cyberpunkrocker-zero, 2021-09-22:
> The situation with character encodings seems to be a mess:

I was afraid that that would be the case. I ended up with the same result after doing a bit of my own research. Well, it is not an earth-shattering issue, just a minor annoyance.

cyberpunkrocker-zero, 2021-09-22:
[https://drive.google.com/file/d/1fcyGoy68dWBvP2di7i18zN-N3s83o-70/view?usp=sharing](url)
Ok, here's a recording which should contain plenty of scandinavian characters in the 'Radio Text'-field

argilo, 2021-09-22:
Thanks! I currently see the following:

[image]

Do you know what the correct characters should be?

cyberpunkrocker-zero, 2021-09-22:
'Kuusi kuvaa eräopaskouluttaja Priitta Pöyhtäri-Trøenin elämästä'
('Six pictures from the life of wilderness guide trainer Priitta Pöyhtäri-Trøen'. Very unconventional surname, half finnish, half norwegian)

argilo, 2021-09-26:
In the recording, the following non-ASCII code points are used:

0x91: ä
0x97: ö
0xf8: ø

The first two match the RDS specification, but the third one does not! According to the specification, ø should be 0xf7. (0xf8 is þ.)

In ISO/IEC 8859-1, ø happens to be represented by 0xf8. Maybe that character was not correctly transcoded before broadcast.

What a mess!

wdnaeng, 2024-03-05:
Agreed, everything about the RDS charset everywhere is a huge mess. I had an Innovonics RDS encoder (very well-regarded broadcast technology company) that had goofily implemented the IEC 62106 charset (table in the pira.cz doc) on its telnet receive side. This was ASCII compatible, but anything beyond that 128 was a mess.

 I was able to get my enyes and vowel diacritics to output OTA correctly (yay multilingual radio) only by forcefeeding the box "wrong-character"-replaced CP1252 so that the thing would read them as the characters I actually wanted.

CP1252:
‚ ƒ „ … † ‡ ˆ ‰ Š š
=>
IEC62106:
é è í ì ó ò ú ù Ñ ñ

I abandoned the project and just replaced them with regular n's and non-diacritic chars because so many actual receivers just didn't bother implementing anything past the first 128 chars anyway, and because I couldn't integrate it into Center Stage Live and would have had to rebuild a lot of its other functionality from scratch.

But maybe this is helpful to somebody somewhere.