gqrx-sdr/gqrx #975: Application is not usable by screenreaders, blind users

hpkotian, 2021-08-31:
I am a blind computer user. I have tried many SDR radio software and regretedly none of them are usable as they do not adhere to accessibility guidelines.
I use a screenreader which is like a middleware by which I can comfortably use the computer. I use Windows10 and the screenreader is NVDA. This a freeware and it can be downloaded from: https://www.nvaccess.org/download/

I hope the GQRX will make the application accessible to us. I see the implementation of QT is not properly implemented. You can find documentation of how to make QT accessible to screenreader on the URL: https://doc.qt.io/qt-5/accessible.html

Presently, I cannot click on the button to start the reciever. Some of the UI elements are read by me. You can check how I use the app by turning off the display, using only the keyboard and by not using the mouse. You should use NVDA for sure for that.

Sure enough, the waterfall will be of no use to me and I have a challenge to make the station selection. Let me list outthe steps
Select the band
Select HF- VHF/UHF etc.
Select modulation WFM, NFM, AM, USB etc.
For step mode- cursor key Up/Down
Auto station selection- Long press cursor Up/Down
Mouse wheel can also be used for manual tuning
I will also need to use the AGC, manual gain control, noise blanker, volume etc.
A way to label the current station which I can get back on a future date.
Ability to key in the station by entering the freq

An easy approach would be to make a fresh app, that would be short sighted. The code would never be updated in future. It should be done on the principal app itself. On the brighter side you will learn to make an accessible application which will ensure all apps made in the future would be directly usable by us and having larger user group. There of course laws which mandates apps to be accessible and it is there where you will find value addition in making the apps accessible complient.
Many users would find the approach better to use and hopeful there would be better ease of use.

I will be happy to work on this.
Harish

xgpt, 2023-03-07:
Following.

hpkotian, 2023-06-27:
Don't see any momentum to my request. Can someone work on it?

xgpt, 2023-06-27:
I'm actually working on a blind user interface for use with not just GQRX but any ham lib compatible radio! (Of which GQRX sort of is!)

It's a separate project from this but was originally dependent on GQRX.