gqrx-sdr/gqrx #980: Startup problem

Section25, 2021-09-25:
I'm using the latest version on a Mac Mini OS X 10.15.7 through a RTL-SDR dongle.

It seemed to work fine for a while but then seemed to hang on startup, I got SBBoD and in the Dock, it said Application not responding and I had to force quit. I rebooted and it was fine for a while but when  tried to use it again after having quit it, I had the same problem again. Frustrated, I tried CubicSDR (which I had never tried before) to try to see if the problem was with the dongle but that launched and ran without a problem. I tried re-installing GQRX but the problem persisted.

I kept trying GQRX but sort of gave up on it but today, I tried again and instead of force quitting I forgot about it and about five minutes later, it suddenly appeared. I wondered if it had had to fix itself somehow after being force quit but when I then quit normally and relaunched it, it took five minutes to start and it does this now consistently. If I reboot my Mac, maybe it will have a normal launch and then revert to five minute launches but I can't reboot again right now just to confirm what I have already experienced.

I should point out that no other application has a launch delay like this. I can't think of any changes I have made to my Mac which could have caused this but I would be grateful for any suggestions.

Section25, 2021-11-07:
It got unbearable so I uninstalled with AppDelete and reinstalled from scratch and it seems to be working as before.

argilo, 2022-01-29:
> it seems to be working as before.

By "working as before", do you mean that it's working properly, or that it still hangs?

Section25, 2022-01-29:
Working properly.

argilo, 2022-01-29:
OK, in that case I'll close this off since there wasn't enough information to investigate this problem and I don't expect you have a way to reproduce it now. If the issue comes up again in the latest version of Gqrx, please let me know and we can reopen this.

closed 2022-01-29 by argilo