gqrx-sdr/gqrx #985: ./macos_bundle.sh

markjfine, 2021-10-04:
Interesting observation:

If you don't have dylibbundler, qt's macdeployqt will add all the dependent frameworks _and dynamic libraries_ into Gqrx.app/Contents/Frameworks. This would normally be fine, but causes some erratic behavior; i.e., crashing when you click the RDS button because it tries to connect to a bad channel.

However, if you manually move all the dylibs from Gqrx.app/Contents/Frameworks to Gqrx.app/Contents/Libs (after it's been  created) everything works as expected. #weird

I should point out that the binary executable works just fine when invoked from the terminal. It's just when building a mac app thing that acts a little wonky if you only rely on qt.

Cheers on releasing 2.14.5! The RDS revision works great!

Running with:
macOS Catalina 10.15.7 on Early 2013 15" MBP
qt 5.15.2
gnuradio 3.9.2.0_2 (from homebrew - you can't cheat and get 3.8 anymore)
gr-osmosdr 2.3.0 (modified to reverse all the boost stuff and disable the python/swig bindings so it builds properly)

argilo, 2021-10-04:
Out of curiosity, what are you using macos_bundle.sh for? I created it for use in Gqrx's automated builds and didn't anticipate other uses.

markjfine, 2021-10-04:
I use it to basically build a quick Gqrx.app from the source tree so I can easily run it from the dock.

This way I can do a 'git pull origin master' (or test branch), cmake/make, then build a fresh app bundle from that (and archive the dmg if needed). Having the app with localized dylibs is kind of handy when homebrew decides I need new supporting software that might break it.

I suppose as an alternative I could just install the executable to /usr/local/bin and create an AppleScript-based app to call it like I do with other things, but this is generally a cleaner process.

argilo, 2021-10-04:
Interesting. I'm glad you find it useful.

I guess to fix the script I should just add the `-e` flag so that the script exits if an error occurs.

markjfine, 2021-10-04:
Or, I could find out why qt is doing what it does and find a way to correct it in the script if dylibbundler isn't found.

markjfine, 2021-10-04:
no way around the qt thing. maybe change the end of the script to something like:

`dbexists=0`
`if [[ -f "/usr/local/bin/dylibbundler" ] || [ -f "/usr/bin/dylibbundler" ]]`
`   dbexists=1`
`fi`
` `
`if [ $dbexists = "1" ]; then`
`   dylibbundler -s /usr/local/opt/icu4c/lib/ -od -b -x Gqrx.app/Contents/MacOS/gqrx -x Gqrx.app/Contents/soapy-modules/libPlutoSDRSupport.so -x Gqrx.app/Contents/soapy-modules/libremoteSupport.so -d Gqrx.app/Contents/Libs/`
`fi`
` `
`ln -s /usr/local/opt/python@3.9/Frameworks/Python.framework /usr/local/opt/python@3.9/lib/Python.framework`
` `
`if [ $dbexists = "0" ]; then`
`   mkdir -p Gqrx.app/Contents/Libs`
`   /usr/local/opt/qt@5/bin/macdeployqt Gqrx.app -no-strip`
`   mv Gqrx.app/Contents/Frameworks/*.dylib Gqrx.app/Contents/Libs`
`fi`
` `
`/usr/local/opt/qt@5/bin/macdeployqt Gqrx.app -dmg -no-strip -always-overwrite`
`mv Gqrx.dmg Gqrx-$GQRX_VERSION.dmg`

I'll test and let you know.

markjfine, 2021-10-04:
This seems to work. Its' an either/or type of thing with macdeployqt because I can run it once to move everything into Libs without building the dmg, but then it puts them right back in Frameworks again when making the dmg. 🤷🏻‍♂️

[macos_bundle.txt](https://github.com/gqrx-sdr/gqrx/files/7279807/macos_bundle.txt)

argilo, 2021-10-04:
The script is meant to use dylibbundler, so in the interest of simplicity I'd like to have it just fail if dylibbundler is not installed, rather than trying to work around the problem. That's what I've done in https://github.com/gqrx-sdr/gqrx/pull/987. Does the change work for you?

markjfine, 2021-10-04:
The -e may be problematic, even if you have dylibbundler, for two reasons:

1. If it doesn't build the soapy-module PlutoSDRSupport it will create an error:
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/otool-classic: can't open file: Gqrx.app/Contents/soapy-modules/libPlutoSDRSupport.so (No such file or directory)
Cannot find file Gqrx.app/Contents/soapy-modules/libPlutoSDRSupport.so to read its dependencies

2. If it gets past that issue, and the link to Python.framework already exists, it will generate a second error:
ln: /usr/local/opt/python@3.9/lib/Python.framework/Python.framework: File exists

I just went ahead and installed dylibbundler, which I really should have just done in the first place. Having said that, since libPlutoSDRSupport.so and libremoteSupport.so aren't on my system, it bails out as well. All I have is librtlsdrSupport.so in /usr/local/lib/SoapySDR/modules0.8.

So if I take those two -x switches out the dylibbundle command line, whole thing runs as intended for me.

Bottom line, perhaps the best course of action for the repository (and for you as well) is to revert the shell script back the way it was and mark this closed.

I can always have a modified script in my local tree, which is good enough for my needs.

argilo, 2021-10-04:
> If it doesn't build the soapy-module PlutoSDRSupport it will create an error

I think this is desirable. I'd want the automated build to fail in that case.

> ln: /usr/local/opt/python@3.9/lib/Python.framework/Python.framework: File exists

I fixed this by adding the `-f` flag to the `ln` command.

PR #987 (2021-10-04): Add -e flag to exit on error
Fixes #985.

/cc @markjfine
closed 2021-10-04 by argilo