gqrx-sdr/gqrx #986: Feature Request: Easy use of usr-defined 2nd audio device

markjfine, 2021-10-04:
Lately have been using Gqrx to do some digital mode stuff. This requires piping output to VB-Cable so a second app can use the audio for decoding purposes, e.g., Dream for DRM.

In order to do this in Gqrx, one needs to:
1. Stop current DSP processing
2. Bring up the Configure IO Devices dialog.
3. Switch the Audio output from Default (or whatever I'm listening on) to VB-Cable and click Ok.
4. Re-engage the DSP processing.
5. Repeat from step one to go back to your primary audio output device.

It works very well, but does make it a little cumbersome to switch back and forth between standard, internally demodulation modes and external processes such as Dream, fldigi, etc.

A nice-to-have feature would be to have one button on the toolbar that re-routes the audio to a second, user-defined audio sink, then routes it back to the primary one once disengaged.

Another option, albeit rather ambitious, would be to add other digital demodulation schemes to Gqrx, such as DRM and NSRC5 for audio, as well as other modes to complement the AFSK tool (such as F6CTE's fabulous Windows-based MultiPSK), but that might be astonomically out of scope for this project.

markjfine, 2021-10-12:
Just thinking ahead:
Since many digital decoding apps don't have a 'passthrough' capability, I wonder if it's also possible to configure it such that you can keep the main audio, but add a route to the second device, kind of like an Aux Out. If so, that might be a good option to have.

effeerre75, 2021-10-16:
At least on Mac you can create a device with multiple out and select it from GQRX.  Here I'm using BlackHole as VB-Cable and also hearing from the SHARP, but I can switch, enable o mute as I like. Just make sure to have drift correction enabled.
[image]

markjfine, 2021-10-16:
That's exactly what I'm looking for... Cheers!

markjfine, 2021-10-16:
So, I was able to set up a Multi-Out pseudo device that includes my MBP's Built-In speakers, USB 5.1 device, and VB-Cable.
Set Gqrx to use the Multi-Out with the main audio through the USB device, which is what I primarily listen through. Changed Dream's input to VB-Cable and it's output to the Built-In speakers.

Just tuned to 15.105, USB, 10kHz wide to test against a DRM signal. If I need to adjust the master volume while Dream is playing, I can mute it from the macOS toolbar. Everything is working precisely as I originally envisioned it.

I just need to test using some of the other digital decoders, but if this works I don't see a problem elsewhere.

Thanks again!

[image]

[image]