gqrx-sdr/gqrx #988: SigMF support

argilo, 2021-10-04:
It would be nice if Gqrx used [SigMF](https://github.com/gnuradio/SigMF) to save metadata when making I/Q recordings. It could also use the metadata (if available) when playing back I/Q recordings.

LongnoseRob, 2021-11-02:
yes that would make handling IQ data much more versatile for both replay of IQ data within GQRX and for out of GQRX analsysis.

devnulling, 2022-02-04:
Bump on this thread. It'd be super to see this functionality in GQRX. 

I guess it's split into two parts of effort:
1) Writing out the extra sigmf meta file on recording (and switching the IQ output extension from .raw to .sigmf-data)
2) Handling the metadata on playback

The 2nd part is a bit more difficult to handle, as SigMF can support several dataformats, and GQRX afaik, handles everything for the most part internally as fc32. 

IMO a good start would be to just support the 1st item as it'd make the recordings output by GQRX a bit easier to work with other programs that use SigMF.

argilo, 2022-02-04:
100% agreed that this can be split into two parts, and that part 1 would be the easier of the two. I'd love to see it as a new feature in 2.16.

Niautanor, 2023-01-14:
From a very preliminary look at the code it looks like everything that would be required for part 1 would be to:

1. change the file extension for the data file in `MainWindow::startIqRecording()` from `.raw` to `.sigmf-data`
2. Also write a metadata file in `MainWindow::startIqRecording()` (which doesn't need to be any more complex than filling a string template with the info that is currently encoded into the file name and then dumping that in a `.sigmf-meta` file)
3. Ensure that we don't create recordings without metadata or metadata without recordings (in case writing to one of the files fails)
4. Potentially adjust playback file filter to also allow `.sigmf-data` files (with the caveat that this will not work in the general case until part 2 but I guess having .raw files that are not in the format that gqrx expects was always a possibility too)

Did I miss something here?

I'm gonna have a go at making a merge request for this. My one question would be if I should hide this with some feature flag (either compile time or as a chooser in the iq recorder dialog).

vladisslav2011, 2023-01-14:
I think, runtime option (a checkbox for example) is better, than compile-time.
For part 2 you may take a look at these PRs: https://github.com/gqrx-sdr/gqrx/pull/1139 https://github.com/gqrx-sdr/gqrx/pull/1013