gqrx-sdr/gqrx #1000: Feature Request: Some means of selecting between two devices

Section25, 2021-11-07:
I have two USB devices, an RTL-SDR and another cheap one which I got off eBay which also identifies as an RTL283UHIDIR both with serial number 00000001.

I have spent hours trying to change the serial number of one of them using a PC but the software of choice, RTLTool, is only available from a Google Drive which will not allow me to download it without being authorised and authorisation has not been forthcoming.

I've tried using other identifiers from the System Profile but this has caused GQRX to fail to recognise anything at all.

I'd like to be able to switch between the two devices without diving into the spaghetti behind my Mac to swap devices.

It this possible? I'd be very grateful.

LongnoseRob, 2021-11-07:
Hello,

Changing the serial-number of the rtl-sdr sticks should also be possible with the `rtl_eeprom` command from thr rtl-sdr package from osmocom.
I dont know if this is also availble on Mac, but it could be worth a try.

After changing the SN of the sticks to e.g. `1234` and `5678`, you could use `rtl=1234` in one application as the device string and `rtl=5678` in the other instance

Section25, 2021-11-07:
Thanks for the tip. I had looked a couple of times at command line options but it is not an area where I am at all comfortable - if I do anything with Terminal, I am following blindly. I tried to follow a couple of suggested methods but got unrecognised command errors.

Anyway, prompted by your message, I tried again and although that actual route took me to a strange land of online Linux emulators which completely lost me and led to a dead end, I did manage to change the serial number of one device sort of following this:

https://www.aeronetworks.ca/2017/11/software-defined-radios.html