gqrx-sdr/gqrx #1005: Can I install gnuradio 3.9 as gqrx's dependency?

TurnOffNOD, 2021-11-22:
Hi, I am installing gqrx on linuxmint latest (based on Ubuntu 20.04 latest), I have added gnuradio's PPA and installed gnuradio 3.9. When I continue to install gqrx, it will always try to installl gnuradio 3.8 as its dependency.
Can I use gnuradio 3.9 instead? Or is there any PPA for newer version for Ubuntu?

argilo, 2021-11-22:
I'm not aware of a maintained PPA that includes Gqrx.

You could either use an AppImage release (https://github.com/gqrx-sdr/gqrx/releases), or else build Gqrx from source.

closed 2021-11-22 by argilo

TurnOffNOD, 2021-11-25:
@argilo
OK.
Just for curiosity, why is gqrx depend on strict version 3.9 of gnuradio?

argilo, 2021-11-25:
Gqrx itself does not depend specifically on GNU Radio 3.9. It can be built against GNU Radio 3.7, 3.8, 3.9 or 3.10.

Each binary version shipped by a Linux distribution will be linked to a specific version of GNU Radio.