gqrx-sdr/gqrx #1014: AppImage does not work

anomalyLuna, 2021-12-12:
I get the following when trying to use the AppImage:
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Gtk-Message: 08:27:53.987: Failed to load module "xapp-gtk3-module"
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
BandPlanFile is /home/anomaly/.config/gqrx/bandplan.csv
BookmarksFile is /home/anomaly/.config/gqrx/bookmarks.csv
Exception in static block reg_basic_and_lf_dboards
  LookupError: KeyError: The dboard id 0x0000 is already registered to Basic TX.
Exception in static block reg_rfx_dboards
  LookupError: KeyError: The dboard id pair [0x0024, 0x0028] is already registered to RFX400.
Exception in static block reg_xcvr2450_dboard
  LookupError: KeyError: The dboard id pair [0x0061, 0x0060] is already registered to XCVR2450.
Exception in static block reg_sbx_dboards
  LookupError: KeyError: The dboard id pair [0x0054, 0x0055] is already registered to SBX.
Exception in static block reg_ubx_dboards
  LookupError: KeyError: The dboard id pair [0x0074, 0x0073] is already registered to UBX v0.3.
Exception in static block reg_wbx_simple_dboards
  LookupError: KeyError: The dboard id pair [0x0053, 0x0052] is already registered to WBX.
Exception in static block reg_dbsrx_dboard
  LookupError: KeyError: The dboard id 0x000d is already registered to DBSRX.
Exception in static block reg_unknown_dboards
  LookupError: KeyError: The dboard id 0xfff0 is already registered to Unknown TX.
Exception in static block reg_tvrx_dboard
  LookupError: KeyError: The dboard id 0x0040 is already registered to TVRX.
Exception in static block reg_dbsrx2_dboard
  LookupError: KeyError: The dboard id 0x0012 is already registered to DBSRX2.
Exception in static block reg_tvrx2_dboard
  LookupError: KeyError: The dboard id 0x0046 is already registered to TVRX2.
Exception in static block reg_twinrx_dboards
  LookupError: KeyError: The dboard id 0x0091 is already registered to TwinRX Rev A.
[ERROR] SoapySDR::loadModule(/tmp/.mount_Gqrx-233jrNe/usr/soapy-modules/libremoteSupport.so)
  duplicate entry for remote (/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libremoteSupport.so)
Found Rafael Micro R820T tuner
[INFO] [UHD] linux; GNU C++ version 8.3.0; Boost_106700; UHD_003.010.003.000-0-unknown
Segmentation fault

I'm using Debian. It looks like some kind of error related to the "SoapySDR" module. Anyone have any suggestions?

vladisslav2011, 2021-12-13:
I get libremoteSupport.so error too, but it works well.
It looks, like you have loaded the UHD library twice and these different libraries are linked against different versions of boost libraries.
Try uninstalling (or just moving out of search path for the time of test) your distribution-installed SoapySDR plugins and give it a try.

akoun, 2021-12-24:
Similar situation here under Debian testing... I just get rhe Exceptions and then it bombs.
thanx in advance for any help
aK
----
> ~/Downloads/Opera/AppImages/Gqrx-2.15.1-x86_64.AppImage 
Exception in static block reg_basic_and_lf_dboards
  LookupError: KeyError: The dboard id 0x0000 is already registered to Basic TX.
Exception in static block reg_rfx_dboards
  LookupError: KeyError: The dboard id pair [0x0024, 0x0028] is already registered to RFX400.
Exception in static block reg_xcvr2450_dboard
  LookupError: KeyError: The dboard id pair [0x0061, 0x0060] is already registered to XCVR2450.
Exception in static block reg_sbx_dboards
  LookupError: KeyError: The dboard id pair [0x0054, 0x0055] is already registered to SBX.
Exception in static block reg_ubx_dboards
  LookupError: KeyError: The dboard id pair [0x0074, 0x0073] is already registered to UBX v0.3.
Exception in static block reg_wbx_simple_dboards
  LookupError: KeyError: The dboard id pair [0x0053, 0x0052] is already registered to WBX.
Exception in static block reg_dbsrx_dboard
  LookupError: KeyError: The dboard id 0x000d is already registered to DBSRX.
Exception in static block reg_unknown_dboards
  LookupError: KeyError: The dboard id 0xfff0 is already registered to Unknown TX.
Exception in static block reg_tvrx_dboard
  LookupError: KeyError: The dboard id 0x0040 is already registered to TVRX.
Exception in static block reg_dbsrx2_dboard
  LookupError: KeyError: The dboard id 0x0012 is already registered to DBSRX2.
Exception in static block reg_tvrx2_dboard
  LookupError: KeyError: The dboard id 0x0046 is already registered to TVRX2.
Exception in static block reg_twinrx_dboards
  LookupError: KeyError: The dboard id 0x0091 is already registered to TwinRX Rev A.

argilo, 2021-12-28:
@anomalyLuna @akoun I recently fixed a startup crash in #1033, and it sounds very similar to this issue. Could you test out this pre-release AppImage and see whether it resolves the crashes you're seeing?

https://github.com/gqrx-sdr/gqrx/suites/4774227554/artifacts/133019782

vladisslav2011, 2021-12-28:
#1033 does not fix this issue.
Conditions: debian buster.
```
./Gqrx-v2.15.1-6-g4a40f72-x86_64.AppImage 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-vlad'
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
gr::fft: can't import wisdom from /home/vlad/.gr_fftw_wisdom
BandPlanFile is /home/vlad/.config/gqrx/bandplan.csv
BookmarksFile is /home/vlad/.config/gqrx/bookmarks.csv
Exception in static block reg_basic_and_lf_dboards
  LookupError: KeyError: The dboard id 0x0000 is already registered to Basic TX.
Exception in static block reg_rfx_dboards
  LookupError: KeyError: The dboard id pair [0x0024, 0x0028] is already registered to RFX400.
Exception in static block reg_xcvr2450_dboard
  LookupError: KeyError: The dboard id pair [0x0061, 0x0060] is already registered to XCVR2450.
Exception in static block reg_sbx_dboards
  LookupError: KeyError: The dboard id pair [0x0054, 0x0055] is already registered to SBX.
Exception in static block reg_ubx_dboards
  LookupError: KeyError: The dboard id pair [0x0074, 0x0073] is already registered to UBX v0.3.
Exception in static block reg_wbx_simple_dboards
  LookupError: KeyError: The dboard id pair [0x0053, 0x0052] is already registered to WBX.
Exception in static block reg_dbsrx_dboard
  LookupError: KeyError: The dboard id 0x000d is already registered to DBSRX.
Exception in static block reg_unknown_dboards
  LookupError: KeyError: The dboard id 0xfff0 is already registered to Unknown TX.
Exception in static block reg_tvrx_dboard
  LookupError: KeyError: The dboard id 0x0040 is already registered to TVRX.
Exception in static block reg_dbsrx2_dboard
  LookupError: KeyError: The dboard id 0x0012 is already registered to DBSRX2.
Exception in static block reg_tvrx2_dboard
  LookupError: KeyError: The dboard id 0x0046 is already registered to TVRX2.
Exception in static block reg_twinrx_dboards
  LookupError: KeyError: The dboard id 0x0091 is already registered to TwinRX Rev A.
[ERROR] SoapySDR::loadModule(/tmp/.mount_Gqrx-vthz0xg/usr/soapy-modules/libremoteSupport.so)
  duplicate entry for remote (/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libremoteSupport.so)
Found Rafael Micro R820T tuner
Found Rafael Micro R820T tuner
Segmentation fault (core dumped)
```
Backtrace:
```
[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_003.010.003.000-0-unknown[New Thread 0x7fffbaffd700 (LWP 3879)]

Thread 1 "AppRun.wrapped" received signal SIGSEGV, Segmentation fault.
0x00007ffff3d5f083 in std::locale::~locale() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff3d5f083 in std::locale::~locale() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff26f1cb2 in uhd::_log::log::~log() () from /tmp/.mount_Gqrx-vvrLliD/usr/bin/../lib/libuhd.so.003.010.003
#2  0x00007fffc3af4e40 in ?? () from /usr/lib/x86_64-linux-gnu/libuhd.so.3.13.1
#3  0x00007fffc3af6dd0 in ?? () from /usr/lib/x86_64-linux-gnu/libuhd.so.3.13.1
#4  0x00007fffc3af6fa1 in ?? () from /usr/lib/x86_64-linux-gnu/libuhd.so.3.13.1
#5  0x00007fffc3b01460 in ?? () from /usr/lib/x86_64-linux-gnu/libuhd.so.3.13.1
#6  0x00007ffff27947ed in uhd::device::find(uhd::device_addr_t const&, uhd::device::device_filter_t) () from /tmp/.mount_Gqrx-vvrLliD/usr/bin/../lib/libuhd.so.003.010.003
#7  0x00007fffd8732ca2 in ?? () from /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libuhdSupport.so
#8  0x00007ffff15e4d13 in SoapySDR::Device::enumerate(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&) ()
   from /tmp/.mount_Gqrx-vvrLliD/usr/bin/../lib/libSoapySDR.so.0.6
#9  0x00007ffff753ffd1 in ?? () from /tmp/.mount_Gqrx-vvrLliD/usr/bin/../lib/libgnuradio-osmosdr.so.0.2.0
#10 0x00007ffff74e8188 in osmosdr::device::find(osmosdr::device_t const&) () from /tmp/.mount_Gqrx-vvrLliD/usr/bin/../lib/libgnuradio-osmosdr.so.0.2.0
#11 0x000055555569259a in CIoConfig::getDeviceList(std::map<QString, QVariant, std::less<QString>, std::allocator<std::pair<QString const, QVariant> > >&) ()
#12 0x0000555555601f8e in MainWindow::MainWindow(QString const&, bool, QWidget*) ()
#13 0x00005555555e1b48 in main ()
```

argilo, 2021-12-29:
I guess one way or another we will need to prevent the AppImage's SoapySDR from loading the system's SoapySDR modules.

PR #1035 (2021-12-30): Prevent SoapySDR from loading incompatible modules
Fixes #1014.

Setting the `SOAPY_SDR_ROOT` to an invalid path prevents the system's SoapySDR modules from being loaded. I've verified that this change allows Gqrx to run on Debian 10 even when the soapysdr0.6-module-uhd package is installed.

A pre-release build is available here: https://github.com/gqrx-sdr/gqrx/suites/4789753922/artifacts/133971350

/cc @anomalyLuna @akoun @vladisslav2011
closed 2021-12-30 by argilo