gqrx-sdr/gqrx #1023: Question: Is it fine to improve performance and decrease CPU usage at high input rates at cost of icreased code complexity?

vladisslav2011, 2021-12-16:
Should we make I/Q swap block switchable out of flowgraph, as it was made to DC remove block to improve the performance.
This block consumes about 5.8% of a CPU core on Core I5-2400 at 20Msps for doing nothing when disabled.
Patching osmosdr_source to swap out it's IQ corrector when it is disabled may give comparable decrease in CPU load (4.9%).
Adding additional modes of operation to DC offset correction block: auto (current mode), measure, apply may give significant improvement when this block is enabled as IIR filer gets heavy at high sample rates.
Moving the DC offset correction function to osmosdr_source block may give even higher improvement in performance as for 8/16bit devices it can be done at zero cost just by altering lookup tables on the fly (measure/apply mode).
Moving RX FFT calculation to dedicated thread should improve waterfall rendering performance on SMP systems at high sample rates/long FFT (1M points). Moving FFT/panadapter picture rendering to separate thread should improve waterfall performance on SMP systems even more. We can set these threads to low priority to leave more CPU cycles to GNU Radio threads on resource-constrained systems and improve GUI responsiveness.
[image]
.

argilo, 2021-12-17:
> Is it fine to improve performance and decrease CPU usage at high input rates at cost of increased code complexity?

I think each case needs to be evaluated to determine whether the performance benefit is worth the additional complexity.

> Should we make I/Q swap block switchable out of flowgraph, as it was made to DC remove block to improve the performance.

Yes. In general, I think that removing unused blocks from the flowgraph is useful.

> Adding additional modes of operation to DC offset correction block

Exposing additional modes to the user would be undesirable, I think. I'd rather have one mode with good performance. I'd also prefer this to be built using GNU Radio blocks. (Perhaps we should use its DC Blocker instead of IIR + Subtract.)

> Moving the DC offset correction function to osmosdr_source block may give even higher improvement

You could always propose that improvement to the maintainers of gr-osmosdr. But in the long term, I'd like to replace Gqrx's gr-osmosdr dependency with gr-soapy (which is built into GNU Radio starting in version 3.9.2).

> Moving RX FFT calculation to dedicated thread should improve waterfall rendering performance on SMP systems at high sample rates/long FFT (1M points). Moving FFT/panadapter picture rendering to [...]

I'd prefer to avoid adding threads to Gqrx, because they are prone to concurrency bugs (deadlocks, race conditions) which are difficult to diagnose. If there is a *substantial* performance benefit to be gained from using threads, then I might consider it, but only if there is no alternative.

argilo, 2021-12-17:
> Moving RX FFT calculation to dedicated thread should improve waterfall rendering performance

Isn't it already in its own thread? Each block that's part of a GNU Radio flowgraph executes in its own thread.

argilo, 2021-12-17:
> Isn't it already in its own thread?

I realized it's not, because the FFT is done in `get_fft_data` rather than the block's `work` function.

vladisslav2011, 2021-12-17:
> Yes. In general, I think that removing unused blocks from the flowgraph is useful.

OK. I'll prepare a PR to demonstrate performance improvement. I've done it already on top of my branch. Should I redo it on top of current master (rebase would not work due to many changes in block connection logic)?

> Exposing additional modes to the user would be undesirable, I think.

I'd rather prefer to have a dropdown with several options than current checkbox. Possible mode options are:
1. None (current checkbox cleared)
2. Software auto (current checkbox set)
3. Software manual. Lock updates of current offsets from moving average/iir filter, switch to add_const_cc to reduce noise?
4. Hardware auto. Call the input device set_dc_offset_mode(2,0) method. Some devices have implemented this in hardware.
5. Hardware manual. Call the input device set_dc_offset_mode(1,0) method. The same as 3 but, maybe, in hardware.
Optional GUI fields to set dc offset values and a button to apply it (call set_dc_offset in "hardware manual" mode or set add_const_cc argument in "software manual" mode).

>  I'd rather have one mode with good performance.

Hardware implementation (or low level driver implementation) may have the best performance, but not every device has it.

>I'd also prefer this to be built using GNU Radio blocks. (Perhaps we should use its DC Blocker instead of IIR + Subtract.)

GNU Radio block (DC Blocker or funny correctiq* blocks, that do the same thing in slightly different way) is universal, but it would never have better performance, than hardware implementation (injected DC from DAC to ADC inputs, like LimeSDR have for example). From my point of view, it is better to leave an option to user to try hardware, that may or may not work and switch to software that would use some CPU time, but work always. And an option to switch to manual mode, removing moving average/iir filter from flowgraph completely.

> But in the long term, I'd like to replace Gqrx's gr-osmosdr dependency with gr-soapy (which is built into GNU Radio starting in version 3.9.2).

gr-soapy may be fine...in distant future. Now "soapy sdr support" option of osmocom_sdr is the thing, that prevents Appimage version from starting (instant segfault) on some systems. See #1014. SoapySDR needs better library loading implementation. At least it should prevent loading a driver that is already loaded (overlapping symbols. Load with RTLD_LOCAL?) and prevent loading driver, linked against different version of boost (instant crash. Inspect symbols like ldd does before loading?). And it should provide an option to disable default driver search paths. Now it only allows to add extra driver search path from an environment variable to builtin path list...

> I'd prefer to avoid adding threads to Gqrx, because they are prone to concurrency bugs...

I completely agree. But... Hmmm. Each GNU Radio block adds a new thread. Avoiding adding threads = avoiding using too much blocks. :-)
I think, we should use dedicated threads only to do computationally heavy tasks, like rx_fft, waterfall/panadapter image rendering, maybe audio_fft too and tasks, that may block in syscalls, like filesystem operations, device list refreshing, maybe initial device enumeration (to show the GUI just after clicking the icon and not after 5-10 seconds, like it is done now).

Current rx_fft implementation is an example of how not to do things in GNU Radio application: it blocks the flowgraph for the long time (at 256k...1M fft sizes and high framerate), using the mutex incorrectly, uses for-loop instead of libvolk function to apply the window, uses boost::circular_buffer instead of excellent double mapped GNU Radio circular buffer, performs heavy task in GUI event processing thread, leading to user interface lags...
I have done some work on rx_fft in https://github.com/vladisslav2011/gqrx/tree/fft_skipping_fix branch. Feel free to checkout and test. But it still needs some work on moving picture rendering to separate thread too. You may zoom the fft in (it decreases rendering time) to test the performance at 1M fft size. The framerate should be higher than in current master. And setting RX fft to 1M points @60FPS does not lead to skipping/lost buffers/lags anymore.

+ I have found another CPU hog: rx_agc_cc uses too much time when the squelch is (sic.) closed! More, than when the squelch is open and audio samples are processed. Should we switch to gated squelch mode to mitigate this issue?
[image]

closed 2022-01-12 by vladisslav2011