gqrx-sdr/gqrx #1027: ppm not working

pidloop, 2021-12-18:
I find that changing the ppm frequency correction value makes no difference in gqrx versions 2.15.1. I loaded version 2.14.5 and that also does nothing but ppm control works fine in 2.11.3.

I run macOS 11.6 and RFSpace CloudIQ. The Device string in all cases is cloudiq=192.168.7.12:50000, input rate 240000. The way I determine the correction is to listen to 10 MHz WWV and adjust ppm until the audio tones are the same in LSB and USB. Without this working, I must tune to 10.000012 MHz to get the pitches to match, changing the ppm value makes no difference. The CloudIQ itself has not changed since I purchased it several years ago.

vladisslav2011, 2021-12-18:
v2.11.3 was using forked gr-osmosdr-gqrx
Current master is using mainline gr-osmosdr.
Mainline gr-osmosdr lacks support of setting frequency correction on RFSpace devices.

pidloop, 2021-12-18:
Sorry to hear this. Hope it can be ported again someday.

argilo, 2021-12-18:
Let's get a patch submitted to gr-osmosdr then. I have submitted several patches in the past which were merged.

Is the forked code available somewhere?

argilo, 2021-12-18:
I guess it's here: https://github.com/csete/gr-osmosdr-gqrx

argilo, 2021-12-20:
And here's the missing commit: https://github.com/csete/gr-osmosdr-gqrx/commit/d66df300eb53df3b153918919c3cefffb1c49051

argilo, 2022-01-23:
I sent the patch to the osmocom-sdr mailing list. Hopefully it will be accepted.

https://lists.osmocom.org/hyperkitty/list/osmocom-sdr@lists.osmocom.org/thread/MY2MEORWKXPFW4KJ7FDY77IBLAQTKSM4/