gqrx-sdr/gqrx #1030: BUG: Freeze after using WAV recorder while playing IQ file

vladisslav2011, 2021-12-27:
Steps to reproduce:

1. Start gqrx
2. Open IQ Tool
3. Select some file
4. Click Play
5. Click Rec in audio dock
6. Click Rec in audio dock once more to stop recording
7. Click Play to stop playback
8. Try to close IQ Tool...

Expected behavior:
gqrx switches to hardware device, IQ Tool gets closed, etc.

Observed behavior:
Freeze

Current git master and appimage are affected.
Fixed in https://github.com/gqrx-sdr/gqrx/pull/1013 (just workaround GNU Radio bug by resetting reciever->rx before calling set_input_rate)
Primary cause: incorrect GNU Radio hier_block2_detail implementation...

argilo, 2021-12-30:
> Primary cause: incorrect GNU Radio hier_block2_detail implementation...

Could you explain this further? Is there a bug in GNU Radio?

vladisslav2011, 2021-12-30:
Yes. There is a bug in the GNU Radio.
When a hier_block2 output is connected to more than one block, disconnecting at least one of connected blocks, but leaving at least one connected block breaks reference to parent detail in hier_block2_detail, making hier_block2::lock() deadlock in loop.
Disconnecting a block from the hier_block2 output clears hier_block2_detail::d_parent_detail
https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/hier_block2_detail.cc#L309-L312
hier_block2::lock() call hier_block2_detail::lock(), which checks hier_block2_detail::d_parent_detail and calls hier_block2_detail::d_owner->lock() (as hier_block2_detail::d_parent_detail is cleared), which calls hier_block2_detail::lock() and so on. 
https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/hier_block2_detail.cc#L873-L881

argilo, 2021-12-30:
OK, in that case you should open an issue (and optionally a pull request) in GNU Radio.

vladisslav2011, 2021-12-30:
I can't reproduce it outside of gqrx.
A simple program, that creates hier_block2 with copy block inside connected to 1 null_source and 2 null_sinks does not freeze when one null_sink gets disconnected. But gqrx freezes in hier_block2::lock() after disconnecting wavfile_sink and trying to stop IQ playback/switch devices. gdb shows main thread stuck jumping between hier_block2::lock and hier_block2_detail::lock with d_parent_detail set to 0.
I have to figure exact conditions, that trigger the bug before opening an issue in GNU Radio tracker.

argilo, 2021-12-30:
> does not freeze when one null_sink gets disconnected

Do you call `lock()` after disconnecting the block?

argilo, 2021-12-30:
The line that triggers the infinite recursion is this one, in `nbrx::set_quad_rate(float)`:

https://github.com/gqrx-sdr/gqrx/blob/d916a202742a026d2d3396a5efd87b6408bca068/src/receivers/nbrx.cpp#L109

```
[...]
#10472 0x00007ffff631255d in gr::hier_block2_detail::lock() (this=0x555555cf03f0) at /home/argilo/prefix_310/src/gnuradio/gnuradio-runtime/lib/hier_block2_detail.cc:880
#10473 0x00007ffff631255d in gr::hier_block2_detail::lock() (this=0x555555cf03f0) at /home/argilo/prefix_310/src/gnuradio/gnuradio-runtime/lib/hier_block2_detail.cc:880
#10474 0x00007ffff631255d in gr::hier_block2_detail::lock() (this=0x555555cf03f0) at /home/argilo/prefix_310/src/gnuradio/gnuradio-runtime/lib/hier_block2_detail.cc:880
#10475 0x00007ffff631255d in gr::hier_block2_detail::lock() (this=0x555555cf03f0) at /home/argilo/prefix_310/src/gnuradio/gnuradio-runtime/lib/hier_block2_detail.cc:880
#10476 0x00005555556c1e88 in nbrx::set_quad_rate(float) (this=0x555555967770, quad_rate=1024000) at /home/argilo/prefix_310/src/gqrx/src/receivers/nbrx.cpp:109
#10477 0x0000555555615f70 in receiver::set_input_rate(double) (this=0x555555d394a0, rate=1024000) at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#10478 0x0000555555619e84 in receiver::set_input_device(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (this=this@entry=0x555555d394a0, device="rtl=0") at /usr/include/c++/9/bits/shared_ptr_base.h:1020
#10479 0x00005555556087a2 in MainWindow::stopIqPlayback() (this=0x7fffffffd750) at /usr/include/c++/9/bits/atomic_base.h:326
#10480 0x00007ffff6b01300 in QMetaObject::activate(QObject*, int, int, void**) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#10481 0x00005555556af40d in CIqTool::on_playButton_clicked(bool) (this=0x555555b95d10, checked=<optimized out>) at /home/argilo/prefix_310/src/gqrx/src/qtgui/iq_tool.cpp:134
[...]
```

vladisslav2011, 2021-12-30:
Thanks.
It looks, like lock() should be called on hier_block2, not on top_block to trigger the bug.

argilo, 2021-12-30:
That's great! A simple reproduction test case will be very helpful to the GNU Radio maintainers.

argilo, 2021-12-30:
I wonder if this bug is what prompted this change (in 2014!):

https://github.com/gqrx-sdr/gqrx/commit/eca0ea7163ff16abd698737fc8ac1baab21be668

argilo, 2021-12-30:
I made a test case for this as well:
```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

#
# SPDX-License-Identifier: GPL-3.0
#
# GNU Radio Python Flow Graph
# Title: Crash Test
# Author: Clayton Smith
# GNU Radio version: 3.10.0.0-rc2

from gnuradio import blocks
from gnuradio import gr

class hier_test(gr.hier_block2):
    def __init__(self):
        gr.hier_block2.__init__(
            self, "Hierarchical Block Test",
            gr.io_signature(1, 1, gr.sizeof_gr_complex*1),
            gr.io_signature(1, 1, gr.sizeof_gr_complex*1),
        )

        self.blocks_copy_0 = blocks.copy(gr.sizeof_gr_complex*1)

        self.connect((self.blocks_copy_0, 0), (self, 0))
        self.connect((self, 0), (self.blocks_copy_0, 0))

class lock_bug_test(gr.top_block):
    def __init__(self):
        gr.top_block.__init__(self, "Bug Test")

        self.hier_test_0 = hier_test()
        self.blocks_vector_source_x_0 = blocks.vector_source_c((0, 0, 0), False, 1, [])
        self.blocks_null_sink_0 = blocks.null_sink(gr.sizeof_gr_complex*1)
        self.blocks_null_sink_1 = blocks.null_sink(gr.sizeof_gr_complex*1)

        self.connect((self.blocks_vector_source_x_0, 0), (self.hier_test_0, 0))
        self.connect((self.hier_test_0, 0), (self.blocks_null_sink_0, 0))
        self.connect((self.hier_test_0, 0), (self.blocks_null_sink_1, 0))

    def trigger_bug(self):
        self.disconnect((self.hier_test_0, 0), (self.blocks_null_sink_1, 0))
        self.hier_test_0.lock()

def main(top_block_cls=lock_bug_test, options=None):
    tb = top_block_cls()
    tb.trigger_bug()

if __name__ == '__main__':
    main()
```

vladisslav2011, 2021-12-30:
I've opened an issue: https://github.com/gnuradio/gnuradio/issues/5436

vladisslav2011, 2021-12-30:
This commit eca0ea7 is not a workaround for the bug. It's workaround for a different issue with the scheduler not knowing what to do with a flowgraph having it's input disconnected (freeze/crash after disconnecting source block from running flowgraph).

argilo, 2021-12-30:
What makes you think that? The commit message ("lock() / unlock() seems to cause occasioanly hangs") sounds like https://github.com/gnuradio/gnuradio/issues/5436 to me.

vladisslav2011, 2021-12-30:
I've spent some time trying to trigger this bug wilt top_block::lock()/top_block::unlock() without success: different combination of nested hier_block2's with different numbers of input/output ports, different order of connection/disconnection...
This bug is triggered only and always by calling hier_block2::lock(), not top_block::lock().
This bug is triggered even when the flowgraph is not running, so stopping the flowgraph does not help.
So, occasional freeze after top_block::lock(), that can be avoided by stopping the flowgraph is completely different issue.

argilo, 2021-12-30:
>So, occasional freeze after top_block::lock(), that can be avoided by stopping the flowgraph is completely different issue.

The commit doesn't just stop the flowgraph, it also removes the `lock()` and `unlock()` calls.

argilo, 2021-12-30:
I guess they are on the top block though, so the fact that you can't reproduce the issue on the top block might suggest something different was happening.

argilo, 2022-01-08:
I made a list of all the hierarchical blocks that use `lock()` and `unlock()`, then checked all the places that those blocks could be disconnected from an external block without a subsequent `connect`. I found two scenarios where this occurs:

1. When audio recording stopped: `receiver::stop_audio_recording()`
2. When the AFSK1200 Decoder window is closed: `receiver::stop_sniffer()`

Both of these scenarios trigger https://github.com/gnuradio/gnuradio/issues/5436, leaving the receiver's `d_parent_detail` in an invalid state. A crash or freeze than then be triggered in at least a few ways:

1. Stop the RDS decoder
2. Change the input device
3. Stop the I/Q player

To work around the bug, we can artificially add extra `connect` calls after the `disconnect` calls in `receiver::stop_audio_recording()` and `receiver::stop_sniffer()` to reset `d_parent_detail`. These artificial calls could then be removed once a fix for https://github.com/gnuradio/gnuradio/issues/5436 is widely available.

argilo, 2022-01-08:
Reopening for now, since the fix was reverted.

argilo, 2022-01-08:
#1036 didn't fully fix this issue; turning off an audio recording and then turning off the RDS receiver would still cause a crash / freeze.

PR #1036 (2021-12-30): Fix #1018
And automatically fix #1030
commit e5e16ccf (2021-12-30): Fix #1018

And automatically fix #1030
PR #1048 (2022-01-08): Revert "Fix #1018"
Reverts gqrx-sdr/gqrx#1036

As discussed in #1040, I've decided to revert this change since it caused regressions. We can fix #1030 another way, and hopefully find a better approach to #1018 as well.
PR #1049 (2022-01-08): Work around GNU Radio issue 5436
Fixes #1030.

This is a workaround for https://github.com/gnuradio/gnuradio/issues/5436. If an external block is disconnected from a hierarchical block, and the hierarchical block is subsequently locked before a new connection is made, then a crash or freeze occurs. We can work around this in Gqrx by adding an extra `connect` call to any place where the bug conditions can be triggered. As far as I can tell, `receiver::stop_audio_recording()` and `receiver::stop_sniffer()` are the only two places where this occurs.

@vladisslav2011 I'd appreciate your help with  [... cut]
closed 2022-01-08 by argilo