gqrx-sdr/gqrx #1032: Segmentation fault (Core dumped) when launch AppImage

ebecamel, 2021-12-27:
Hi all,
I use the version 2.13.4 of the AppImage and I don't have problem, but when I want launch the version 2.14.X or 2.15.X I have a core dumped without error message.
I use Ubuntu 20.04
Do you have an idea ?

dranch, 2021-12-27:
Gqrx 2.15.1 works fine for me on u20.04 with an e4000 based RTL dongle (NooElec SmartTee).  Try deleting your ~/.config/gqrx/default.conf file and try again.

ebecamel, 2021-12-27:
Thanks for your answer but I still have the problem

argilo, 2021-12-27:
Could you please run Gqrx inside gdb and post a stack trace of the crash? To do that:

1. Run `gdb ./Gqrx-2.15.1-x86_64.AppImage` at the command line.
2. Type `run` to start Gqrx.
3. When the program crashes, type `bt` to print the stack trace.

argilo, 2021-12-27:
Also, could you please isolate the exact version where the problem started? Please try each version after 2.13.4 and let me know the earliest one that has the problem.

dranch, 2021-12-27:
It should be noted that I've seen a 50% appimage file size reduction recently across the recent versions.  Not sure if this is expected:

-rwx------ 1 dranch dranch 62721952 Feb 16  2021 Gqrx-2.11.5-x86_64.AppImage
-rwx------ 1 dranch dranch 38184128 Dec 18 09:14 Gqrx-2.15.1-x86_64.AppImage

ebecamel, 2021-12-27:
> Also, could you please isolate the exact version where the problem started? Please try each version after 2.13.4 and let me know the earliest one that has the problem.

You can find here the result when I run the appimage in gbd : 

enzo@ThinkPad-E14-Gen-2:~$ gdb ./Gqrx-2.15.1-x86_64.AppImage
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Gqrx-2.15.1-x86_64.AppImage...
(No debugging symbols found in ./Gqrx-2.15.1-x86_64.AppImage)
(gdb) run
Starting program: /home/enzo/Gqrx-2.15.1-x86_64.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 2937]
process 2933 is executing new program: /usr/bin/env
process 2933 is executing new program: /usr/bin/bash
[Detaching after fork from child process 2942]
process 2933 is executing new program: /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/gqrx
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x555500000001) at malloc.c:3102
3102	malloc.c: Aucun fichier ou dossier de ce type.
(gdb) bt
#0  __GI___libc_free (mem=0x555500000001) at malloc.c:3102
#1  0x00007ffff3d7e2af in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007fffef76a1c1 in boost::filesystem::detail::directory_iterator_construct(boost::filesystem::directory_iterator&, boost::filesystem::path const&, boost::system::error_code*) ()
   from /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/../lib/libboost_filesystem.so.1.65.1
#3  0x00007fffe98130ec in ?? () from /lib/x86_64-linux-gnu/libuhd.so.3.15.0
#4  0x00007fffe98131c0 in ?? () from /lib/x86_64-linux-gnu/libuhd.so.3.15.0
#5  0x00007fffe9820e4e in _uhd_static_fixture::_uhd_static_fixture(void (*)(), char const*) ()
   from /lib/x86_64-linux-gnu/libuhd.so.3.15.0
#6  0x00007ffff7fe0b8a in ?? () from /lib64/ld-linux-x86-64.so.2
#7  0x00007ffff7fe0c91 in ?? () from /lib64/ld-linux-x86-64.so.2
#8  0x00007ffff3a42915 in __GI__dl_catch_exception (exception=<optimized out>, operate=<optimized out>, 
    args=<optimized out>) at dl-error-skeleton.c:182
#9  0x00007ffff7fe50bf in ?? () from /lib64/ld-linux-x86-64.so.2
#10 0x00007ffff3a428b8 in __GI__dl_catch_exception (exception=<optimized out>, operate=<optimized out>, 
    args=<optimized out>) at dl-error-skeleton.c:208
#11 0x00007ffff7fe45fa in ?? () from /lib64/ld-linux-x86-64.so.2
#12 0x00007ffff106634c in dlopen_doit (a=a@entry=0x7fffffffdaf0) at dlopen.c:66
#13 0x00007ffff3a428b8 in __GI__dl_catch_exception (exception=exception@entry=0x7fffffffda90, operate=<optimized out>, 
    args=<optimized out>) at dl-error-skeleton.c:208
#14 0x00007ffff3a42983 in __GI__dl_catch_error (objname=0x555555899390, errstring=0x555555899398, 
    mallocedp=0x555555899388, operate=<optimized out>, args=<optimized out>) at dl-error-skeleton.c:227
#15 0x00007ffff1066b59 in _dlerror_run (operate=operate@entry=0x7ffff10662f0 <dlopen_doit>, 
    args=args@entry=0x7fffffffdaf0) at dlerror.c:170
#16 0x00007ffff10663da in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
#17 0x00007ffff28720e0 in ?? () from /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/../lib/libuhd.so.003.010.003
#18 0x00007ffff28721cc in ?? () from /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/../lib/libuhd.so.003.010.003
#19 0x00007ffff28725b8 in ?? () from /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/../lib/libuhd.so.003.010.003
#20 0x00007ffff287e8b9 in _uhd_static_fixture::_uhd_static_fixture(void (*)(), char const*) ()
--Type <RET> for more, q to quit, c to continue without paging--

The problem start at the 2.14.2 (2.14.1 work, 2.14.2 give core dumped)

argilo, 2021-12-27:
It looks like only the first page of the stack trace is there. Could you press Enter until the whole stack trace is displayed? Thanks.

> The problem start at the 2.14.2

This is the first version where the AppImage was generated automatically using GitHub Actions. My guess would be that one of the dependencies changed versions.

argilo, 2021-12-27:
Version 2.14.1 was built on Ubuntu 20.04, which ships with UHD 3.15.0.

Version 2.14.2 and later are built on Ubuntu 18.04 so that the AppImage will run on older Linux distributions. Ubuntu 18.04 ships with UHD 3.10.3.

From the stack trace, it looks like the AppImage's UHD (version 3.10.3) is trying to open your system's UHD (version 3.15.0) which causes the crash.

ebecamel, 2021-12-27:
You can find all the gdb run :

enzo@ThinkPad-E14-Gen-2:~$ gdb ./Gqrx-2.15.1-x86_64.AppImage
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Gqrx-2.15.1-x86_64.AppImage...
(No debugging symbols found in ./Gqrx-2.15.1-x86_64.AppImage)
(gdb) run
Starting program: /home/enzo/Gqrx-2.15.1-x86_64.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 2937]
process 2933 is executing new program: /usr/bin/env
process 2933 is executing new program: /usr/bin/bash
[Detaching after fork from child process 2942]
process 2933 is executing new program: /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/gqrx
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x555500000001) at malloc.c:3102
3102	malloc.c: Aucun fichier ou dossier de ce type.
(gdb) bt
#0  __GI___libc_free (mem=0x555500000001) at malloc.c:3102
#1  0x00007ffff3d7e2af in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_assign(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x00007fffef76a1c1 in boost::filesystem::detail::directory_iterator_construct(boost::filesystem::directory_iterator&, boost::filesystem::path const&, boost::system::error_code*) ()
   from /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/../lib/libboost_filesystem.so.1.65.1
#3  0x00007fffe98130ec in ?? () from /lib/x86_64-linux-gnu/libuhd.so.3.15.0
#4  0x00007fffe98131c0 in ?? () from /lib/x86_64-linux-gnu/libuhd.so.3.15.0
#5  0x00007fffe9820e4e in _uhd_static_fixture::_uhd_static_fixture(void (*)(), char const*) ()
   from /lib/x86_64-linux-gnu/libuhd.so.3.15.0
#6  0x00007ffff7fe0b8a in ?? () from /lib64/ld-linux-x86-64.so.2
#7  0x00007ffff7fe0c91 in ?? () from /lib64/ld-linux-x86-64.so.2
#8  0x00007ffff3a42915 in __GI__dl_catch_exception (exception=<optimized out>, operate=<optimized out>, 
    args=<optimized out>) at dl-error-skeleton.c:182
#9  0x00007ffff7fe50bf in ?? () from /lib64/ld-linux-x86-64.so.2
#10 0x00007ffff3a428b8 in __GI__dl_catch_exception (exception=<optimized out>, operate=<optimized out>, 
    args=<optimized out>) at dl-error-skeleton.c:208
#11 0x00007ffff7fe45fa in ?? () from /lib64/ld-linux-x86-64.so.2
#12 0x00007ffff106634c in dlopen_doit (a=a@entry=0x7fffffffdaf0) at dlopen.c:66
#13 0x00007ffff3a428b8 in __GI__dl_catch_exception (exception=exception@entry=0x7fffffffda90, operate=<optimized out>, 
    args=<optimized out>) at dl-error-skeleton.c:208
#14 0x00007ffff3a42983 in __GI__dl_catch_error (objname=0x555555899390, errstring=0x555555899398, 
    mallocedp=0x555555899388, operate=<optimized out>, args=<optimized out>) at dl-error-skeleton.c:227
#15 0x00007ffff1066b59 in _dlerror_run (operate=operate@entry=0x7ffff10662f0 <dlopen_doit>, 
    args=args@entry=0x7fffffffdaf0) at dlerror.c:170
#16 0x00007ffff10663da in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
#17 0x00007ffff28720e0 in ?? () from /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/../lib/libuhd.so.003.010.003
#18 0x00007ffff28721cc in ?? () from /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/../lib/libuhd.so.003.010.003
#19 0x00007ffff28725b8 in ?? () from /tmp/.mount_Gqrx-2BDZ2MX/usr/bin/../lib/libuhd.so.003.010.003
#20 0x00007ffff287e8b9 in _uhd_static_fixture::_uhd_static_fixture(void (*)(), char const*) ()
--Type <RET> for more, q to quit, c to continue without paging--
  Z2MX/usr/bin/../lib/libuhd.so.003.010.003
#21 0x00007ffff7fe0b8a in ?? () from /lib64/ld-linux-x86-64.so.2
#22 0x00007ffff7fe0c91 in ?? () from /lib64/ld-linux-x86-64.so.2
#23 0x00007ffff7fd013a in ?? () from /lib64/ld-linux-x86-64.so.2
#24 0x0000000000000001 in ?? ()
#25 0x00007fffffffe295 in ?? ()
#26 0x0000000000000000 in ?? ()
(gdb) 

For the problem of UHD, how I can solve the problem ?

argilo, 2021-12-27:
> For the problem of UHD, how I can solve the problem ?

You could try `sudo apt remove libuhd3.15.0` to remove the UHD package from your system.

dranch, 2021-12-27:
What SDR are you using?  Are you using a HackRF?  I assume your machine is fully updated (kernel & packages) and rebooted to actually use them.  One additional thing that's different from your machine and mine is you're using the French language pack and I'm using English.   I've seen several unicode related issues in other projects but I know that switching languages might be difficult for you.  To Argilo's last statement, I do NOT have libuhd installed on my system.  Beyond all that speculation from me, you might need a non-optimized Appimage here to get a useful coredump to know what is really breaking.

vladisslav2011, 2021-12-27:
I've reproduced this issue!
Could you try this command `sudo apt remove uhd-soapysdr` and then run gqrx Appimage?

argilo, 2021-12-27:
Thanks for tracking this down @vladisslav2011. I'm also able to reproduce on my system when `uhd-soapysdr` is installed. I guess the AppImage should be configured so that its SoapySDR doesn't try to load modules from outside the AppImage.

vladisslav2011, 2021-12-27:
Not only SoapySDR, but libuhd too should be configured (patched?) to not try to load modules from outside the Appimage.
This time libuhd is loading a module, that brings support of SoapySDR devices to libuhd, that loads distribution libuhd, that loads distribution libboost_filesystem -> crash.

argilo, 2021-12-27:
Right, I guess it is libuhd that's the problem. SoapySDR already has its module directory reconfigured here:

https://github.com/gqrx-sdr/gqrx/blob/9021996c0c4ba771152d15fd3318c0834af45ad8/src/applications/gqrx/main.cpp#L62-L65

libuhd can be configured using an environment variable, but the system path is always searched as well:

https://github.com/EttusResearch/uhd/blob/fc864f36758b08f0c90970e5b54433667b1f038c/host/lib/utils/paths.cpp#L311-L324

The system path compiled into libuhd does not work nicely with AppImages, so I guess it needs to be rebuild from source or else binary-patched.

vladisslav2011, 2021-12-27:
SoapySDR does not support full plugin search path reconfiguration. It just adds additional paths from environment variable:
```
std::vector<std::string> SoapySDR::listSearchPaths(void)
{
    //the default search path
    std::vector<std::string> searchPaths;
    searchPaths.push_back(SoapySDR::getRootPath() + "/@CMAKE_INSTALL_LIBDIR@/SoapySDR/modules" + SoapySDR::getABIVersion());

    //support /usr/local module installs when the install prefix is /usr
    if (SoapySDR::getRootPath() == "/usr")
    {
        searchPaths.push_back("/usr/local/@CMAKE_INSTALL_LIBDIR@/SoapySDR/modules" + SoapySDR::getABIVersion());
        //when using a multi-arch directory, support single-arch path as well
        static const std::string libdir("@CMAKE_INSTALL_LIBDIR@");
        if (libdir.find("lib/") == 0) //startswith lib/ indicating multi-arch
            searchPaths.push_back("/usr/local/lib/SoapySDR/modules" + SoapySDR::getABIVersion());
    }

    //separator for search paths
    #ifdef _WIN32
    static const char sep = ';';
    #else
    static const char sep = ':';
    #endif

    //check the environment's search path
    std::stringstream pluginPaths(getEnvImpl("SOAPY_SDR_PLUGIN_PATH"));
    std::string pluginPath;
    while (std::getline(pluginPaths, pluginPath, sep))
    {
        if (pluginPath.empty()) continue;
        searchPaths.push_back(pluginPath);
    }

    return searchPaths;
}
```
Setting SOAPY_SDR_PLUGIN_PATH just allows us to load SoapySDR modules from inside the Appimage, but does not prevent it from searching paths, configured at build time.
If Appimage version is built using distribution supplied SoapySDR, then there might be a problem... If SoapySDR is built from source and installed outside of /usr as a part of gqrx Appimage build process, then it's OK.

argilo, 2021-12-27:
Setting the `UHD_PKG_PATH` environment variable allows Gqrx to start without crashing:

```
UHD_PKG_PATH=foobar ./Gqrx-2.15.1-x86_64.AppImage
```

vladisslav2011, 2021-12-27:
Yes. Setting UHD_PKG_PATH to some invalid path prevents libuhd from searching other paths and fixes the issue.
https://github.com/EttusResearch/uhd/blob/fc864f36758b08f0c90970e5b54433667b1f038c/host/lib/utils/load_modules.cpp#L63-L86

argilo, 2021-12-27:
Perhaps something like this would be a good fix?
```diff
diff --git a/src/applications/gqrx/main.cpp b/src/applications/gqrx/main.cpp
index 68059d0cb4..d15b53b9ff 100644
--- a/src/applications/gqrx/main.cpp
+++ b/src/applications/gqrx/main.cpp
@@ -62,8 +62,13 @@ int main(int argc, char *argv[])
     QString plugin_path = QDir::cleanPath(QCoreApplication::applicationDirPath() + "/../soapy-modules");
     QFileInfo plugin_path_info(plugin_path);
     if (plugin_path_info.isDir())
+    {
         qputenv("SOAPY_SDR_PLUGIN_PATH", plugin_path.toUtf8());
 
+        QString uhd_pkg_path = QDir::cleanPath(QCoreApplication::applicationDirPath() + "/../lib");
+        qputenv("UHD_PKG_PATH", uhd_pkg_path.toUtf8());
+    }
+
     // setup controlport via environment variables
     // see http://lists.gnu.org/archive/html/discuss-gnuradio/2013-05/msg00270.html
     // Note: tried using gr::prefs().save() but that doesn't have effect until the next time
```

argilo, 2021-12-27:
Actually, that might not work if the static fixture executes before `main`. :thinking:

vladisslav2011, 2021-12-27:
We can set environment variables using apprun hooks:
```
diff --git a/appimage.sh b/appimage.sh
index 5ef6076..2749960 100755
--- a/appimage.sh
+++ b/appimage.sh
@@ -69,6 +69,9 @@ chmod a+x *.AppImage
 mkdir -p ./AppDir/usr/lib
 cp -R /usr/lib/x86_64-linux-gnu/SoapySDR/modules* ./AppDir/usr/soapy-modules
 
+mkdir -p ./AppDir/apprun-hooks
+echo -ne "export UHD_PKG_PATH="'"'"$PWD/../lib"'"'"\n" >./AppDir/apprun-hooks/uhd-hook.sh
+
 ./linuxdeploy-x86_64.AppImage -e "$APP" -d "$DESKTOP" -i "$ICON" -p qt --output appimage --appdir=./AppDir
 RESULT=$?
 
```

ebecamel, 2021-12-27:
The solution of @vladisslav2011 work ! After `sudo apt remove uhd-soapysdr` it's working, thank you !
To answer to @dranch I use a RTL-SDR V3.

PR #1033 (2021-12-28): Set UHD package path in AppImage
Fixes #1032.

By pointing the `UHD_PKG_PATH` environment variable to the AppImage's mount point, UHD modules will be loaded from there instead of from the system (where they may be linked to an incompatible version of libuhd).
closed 2021-12-28 by argilo