gqrx-sdr/gqrx #1039: Regression: some receiver settings does not get restored when switching from IQ Tool to hardware device

vladisslav2011, 2022-01-01:
...or when switching input rates/toggling DC remove.
Introduced in e5e16ccff5e7b50f2f9c50b222819e198a867e5d.
I think, we should store rx filter parameters, cw offset and SQL level to private class members on change and restore when the demodulator gets recreated.
And we should track and restore state of RDS decoder too to prevent a freeze.
PR in progress...

argilo, 2022-01-01:
Couldn't we just use something like loadConfig to load all the settings?

vladisslav2011, 2022-01-01:
No.
`loadConfig` does not restore required settings in this case. `loadConfig` is called when switching input rates/changing input devices, but SQL level and filter low/high cut frequencies still get reset to default. The GUI shows everything to be correct, but demodulator state goes out of sync.

vladisslav2011, 2022-01-01:
PR is ready https://github.com/gqrx-sdr/gqrx/pull/1040

argilo, 2022-01-01:
It would probably be worth adding those settings into the settings file. That would prevent them from being lost when Gqrx is closed, and would remove the need to add yet more complexity to the I/Q tool.

argilo, 2022-01-01:
In any case, that can go in a separate pull request. I'll have a look at this one when I get a chance.

vladisslav2011, 2022-01-01:
These settings are already added to the settings file and restored correctly when gqrx is closed and then restarted. This is different issue.

vladisslav2011, 2022-01-01:
That's not enough.
We should copy NFM max deviation, deemphasis, agc settings, AM dc reduction too.
Adding settings-copying constructor/factory to the `receiver_base` class and tracking setting values in `receiver_base` private fields  may be better solution.

vladisslav2011, 2022-01-01:
PR redone. All settings are restored correctly now.

vladisslav2011, 2022-01-03:
It looks, like this is not the regression.
It's just separate bug, that was made to occur more often after https://github.com/gqrx-sdr/gqrx/commit/e5e16ccff5e7b50f2f9c50b222819e198a867e5d.
It can be reproduced by following steps:
1. Set AM DCR to disabled. The audio spectrum should show DC spike at center.
2. Switch to WFM
3. Switch back to AM

Expected result:
DC spike on the audio spectrum and AM DCR setting disabled.

Observed result:
No DC spike on the audio spectrum, but AM DCR setting disabled.

vladisslav2011, 2022-01-03:
> It would probably be worth adding those settings into the settings file.

> In any case, that can go in a separate pull request.

PR is ready: https://github.com/gqrx-sdr/gqrx/pull/1043