gqrx-sdr/gqrx #1045: Odd crash on losing mouse focus on Wayland

AbortRetryFail, 2022-01-04:
I'm not sure if this is a gqrx bug or a Qt bug, but gqrx 2.15.1 crashes _sometimes_ when losing mouse focus across multiple monitors when Qt has QT_QPA_PLATFORM=wayland set. Running with QT_QPA_PLATFORM=xcb under XWayland works and does not crash in this way.

```
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.4.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
gr::log :WARN: file_source0 - file size is not a multiple of item size
Resampling audio 96000 -> 48000
BandPlanFile is /home/arf/.config/gqrx/bandplan.csv
BookmarksFile is /home/arf/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 11.1.0; Boost_107800; UHD_4.1.0.5-0-unknown
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.4.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T/2 tuner
Allocating 15 zero-copy buffers
[destroyed object]: error 1: xdg_surface must have a role
The Wayland connection experienced a fatal error: Protocol error
```

I have not been able to find anything useful about this `error 1: xdg_surface must have a role` error. Compositor is Sway v1.6.1 with 3 displays configured. The crash always seems to happen when gqrx gets focus, then loses focus as I move my mouse to a different monitor. The About Qt dialog box says it's using Qt 5.15.2. I got gqrx and all of its dependencies from the Archlinux repos.

AbortRetryFail, 2022-01-04:
It's worth noting, other Qt apps on my system don't do this. Some examples are Telegram Desktop, Okular, WSJT-X.

argilo, 2022-02-09:
Thanks for the report. Could you try running Gqrx within gdb and grab a backtrace (with the `bt` command) when a crash occurs? That would be very helpful for tracking down the source of this issue.

AbortRetryFail, 2022-02-12:
Okay I did that, it says no stack.
```
xdg_surface@40: error 1: xdg_surface must have a role
The Wayland connection experienced a fatal error: Protocol error
The Wayland connection experienced a fatal error: Protocol error
[Thread 0x7fff857d1640 (LWP 5677) exited]
[Thread 0x7fff7cfd0640 (LWP 5676) exited]
[Thread 0x7fff84fd0640 (LWP 5675) exited]
[Thread 0x7fff21ffb640 (LWP 5549) exited]
[Thread 0x7fff227fc640 (LWP 5548) exited]
[Thread 0x7fff22ffd640 (LWP 5547) exited]
[Thread 0x7fff237fe640 (LWP 5546) exited]
[Thread 0x7fff23fff640 (LWP 5545) exited]
[Thread 0x7fff38ff9640 (LWP 5544) exited]
[Thread 0x7fff397fa640 (LWP 5543) exited]
[Thread 0x7fff39ffb640 (LWP 5542) exited]
[Thread 0x7fff3a7fc640 (LWP 5541) exited]
[Thread 0x7fff3affd640 (LWP 5540) exited]
[Thread 0x7fff3b7fe640 (LWP 5539) exited]
[Thread 0x7fff3bfff640 (LWP 5538) exited]
[Thread 0x7fff5cff9640 (LWP 5537) exited]
[Thread 0x7fff5d7fa640 (LWP 5536) exited]
[Thread 0x7fff5dffb640 (LWP 5535) exited]
[Thread 0x7fff5e7fc640 (LWP 5534) exited]
[Thread 0x7fff5f7fe640 (LWP 5532) exited]
[Thread 0x7fff5ffff640 (LWP 5531) exited]
[Thread 0x7fff7d7fa640 (LWP 5530) exited]
[Thread 0x7fff7dffb640 (LWP 5529) exited]
[Thread 0x7fff7e7fc640 (LWP 5528) exited]
[Thread 0x7fff7effd640 (LWP 5527) exited]
[Thread 0x7fff7f7fe640 (LWP 5526) exited]
[Thread 0x7fff7ffff640 (LWP 5525) exited]
[Thread 0x7fffedc14640 (LWP 5519) exited]
[Thread 0x7fffb0b2f640 (LWP 5518) exited]
[Thread 0x7fffc1ffb640 (LWP 5517) exited]
[Thread 0x7fffc2ffd640 (LWP 5516) exited]
[Thread 0x7fffc37fe640 (LWP 5515) exited]
[Thread 0x7fffc3fff640 (LWP 5514) exited]
[Thread 0x7fffe4ac6640 (LWP 5513) exited]
[Thread 0x7fffe52c7640 (LWP 5512) exited]
[Thread 0x7fffb2ffd640 (LWP 5511) exited]
[Thread 0x7fffb37fe640 (LWP 5510) exited]
[Thread 0x7fffe5ffd640 (LWP 5509) exited]
[Thread 0x7fffc0ff9640 (LWP 5508) exited]
[Thread 0x7fffb3fff640 (LWP 5507) exited]
[Thread 0x7fffc17fa640 (LWP 5496) exited]
[Thread 0x7fffc27fc640 (LWP 5494) exited]
[Thread 0x7fffef43f640 (LWP 5483) exited]
[Thread 0x7fffefc40640 (LWP 5481) exited]
[Thread 0x7ffff0441640 (LWP 5480) exited]
[Thread 0x7ffff0a6acc0 (LWP 5476) exited]
[Inferior 1 (process 5476) exited with code 01]
(gdb) bt
No stack.
```

QuantMint, 2022-05-21:
Same here, on sway 1.7 NixOS unstable

argilo, 2022-05-26:
@QuantMint Are you able to collect any useful debugging information (with gdb or otherwise)?

argilo, 2022-05-26:
Also, could you please check Help -> About Qt to see which version of Qt is in use?

QuantMint, 2022-05-27:
> @QuantMint Are you able to collect any useful debugging information (with gdb or otherwise)?

Here: [backtrace.txt](https://github.com/gqrx-sdr/gqrx/files/8787577/backtrace.txt)
I am using QT 5.15.3

Tungsten842, 2024-03-17:
I believe this can be closed as I can no longer reproduce the issue anymore.