gqrx-sdr/gqrx #1046: Close the remote control socket even after a hardware disconnect

LongnoseRob, 2022-01-07:
I run GQRX on Linux built from source (git  master) and mostly with remote control active controlled by gpredict.
Today I had an issue with the USB conenction on my airspy-mini and noticed that the sprectvum view was frozen.
So I stopped the DSP and closed GQRX, fixed my USB problem  and restarted GQRX.
The DSP worked fine and I could see the spectrum again and also had audio, so I activated the remote control, but it would not work.
After closing GQRX, i checked with `lsof -i 46`, that the socket on port 7356 was still shown as associated to the old GQRX PID.

I would expect that if I close GQRX in such a stuation (decoder stopped even after hw-failure), all sockets/ports wouls also be properly closed.

Unfotunately I did not take a screenshot or captured data from the shell, but this could be easiealy be reproduced.

vladisslav2011, 2022-01-07:
I can't reproduce. The listening socket always gets closed. It gets correctly closed even when there is an active connection and I kill gqrx with 'KILL' signal making it impossible to do shutdown cleanup.

>So I stopped the DSP and closed GQRX, fixed my USB problem and restarted GQRX.

There is no need to restart gqrx in case of USB error (at least with gr-osmosdr devices). Just replug the device, open the "I/O devices" dialog and click OK. It should continue to work even without restarting a DSP.

LongnoseRob, 2022-01-16:
DId not have the problem anymore, so closing this issue

closed 2022-01-16 by LongnoseRob