gqrx-sdr/gqrx #1047: gqrx stops working on Raspberry Pi

ernstz, 2022-01-07:
I loaded gqrx to a RPi4 connected to a nooelec SDR.
The sofware works fine, but if I try to execute to program for a longer time (some hours), it is stops working.

Maybe this problem is related to issue #916.

vladisslav2011, 2022-01-07:
I do not have a RPi4 unfortunately... But I did some tests on a weak laptop (Intel Atom N270) - it worked for several hours without a freeze.
Could you give some more information, please?
Linux version `uname -a`, libraries versions `ldd gqrx`, GUI backend used (Xorg, Wayland, etc).
Could you describe the exact "non-working" state: no sound, straight lines on the waterfall, no reaction to mouse clicks/hotkeys, please?
In case of complete freeze, could you attach to it (in non-working state) with gdb `attach <PID of gqrx>` an post a backtrace `bt` here?

ernstz, 2022-01-07:
Thanks for your answer and your support!

- Linux version: uname -a
`Linux raspberrypi 5.10.89-v7l+ #1508 SMP Tue Jan 4 19:52:01 GMT 2022 armv7l GNU/Linux`

- Linux version: cat /etc/os-release

```
	PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
	NAME="Raspbian GNU/Linux"
	VERSION_ID="11"
	VERSION="11 (bullseye)"
	VERSION_CODENAME=bullseye
	ID=raspbian
	ID_LIKE=debian
	HOME_URL="http://www.raspbian.org/"
	SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
	BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

- ldd gqrx

```
	pi@raspberrypi:/bin $ ldd gqrx
	linux-vdso.so.1 (0xbef4a000)
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0xb6f2d000)
	libQt5Network.so.5 => /lib/arm-linux-gnueabihf/libQt5Network.so.5 (0xb6d70000)
	libQt5Svg.so.5 => /lib/arm-linux-gnueabihf/libQt5Svg.so.5 (0xb6d14000)
	libgnuradio-osmosdr.so.0.2.0 => /lib/arm-linux-gnueabihf/libgnuradio-osmosdr.so.0.2.0 (0xb6c2f000)
	libpulse.so.0 => /lib/arm-linux-gnueabihf/libpulse.so.0 (0xb6bd4000)
	libpulse-simple.so.0 => /lib/arm-linux-gnueabihf/libpulse-simple.so.0 (0xb6bc0000)
	libgnuradio-digital.so.3.8.2 => /lib/arm-linux-gnueabihf/libgnuradio-digital.so.3.8.2 (0xb6a8d000)
	libQt5Widgets.so.5 => /lib/arm-linux-gnueabihf/libQt5Widgets.so.5 (0xb64c2000)
	libQt5Gui.so.5 => /lib/arm-linux-gnueabihf/libQt5Gui.so.5 (0xb5f19000)
	libQt5Core.so.5 => /lib/arm-linux-gnueabihf/libQt5Core.so.5 (0xb5a36000)
	libgnuradio-analog.so.3.8.2 => /lib/arm-linux-gnueabihf/libgnuradio-analog.so.3.8.2 (0xb59c0000)
	libgnuradio-filter.so.3.8.2 => /lib/arm-linux-gnueabihf/libgnuradio-filter.so.3.8.2 (0xb5910000)
	libgnuradio-blocks.so.3.8.2 => /lib/arm-linux-gnueabihf/libgnuradio-blocks.so.3.8.2 (0xb5690000)
	libgnuradio-fft.so.3.8.2 => /lib/arm-linux-gnueabihf/libgnuradio-fft.so.3.8.2 (0xb5658000)
	libgnuradio-runtime.so.3.8.2 => /lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.8.2 (0xb54ff000)
	libvolk.so.2.4 => /lib/arm-linux-gnueabihf/libvolk.so.2.4 (0xb548f000)
	libgnuradio-pmt.so.3.8.2 => /lib/arm-linux-gnueabihf/libgnuradio-pmt.so.3.8.2 (0xb5430000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb5404000)
	liblog4cpp.so.5 => /lib/arm-linux-gnueabihf/liblog4cpp.so.5 (0xb53c2000)
	libstdc++.so.6 => /lib/arm-linux-gnueabihf/libstdc++.so.6 (0xb523a000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb51cb000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0xb519e000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb504a000)
	/lib/ld-linux-armhf.so.3 (0xb6f42000)
	libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0xb5022000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0xb500e000)
	libgssapi_krb5.so.2 => /lib/arm-linux-gnueabihf/libgssapi_krb5.so.2 (0xb4fbe000)
	libboost_chrono.so.1.71.0 => /lib/arm-linux-gnueabihf/libboost_chrono.so.1.71.0 (0xb4fa6000)
	libboost_thread.so.1.71.0 => /lib/arm-linux-gnueabihf/libboost_thread.so.1.71.0 (0xb4f76000)
	libgnuradio-iqbalance.so.3.8.0 => /lib/arm-linux-gnueabihf/libgnuradio-iqbalance.so.3.8.0 (0xb4f52000)
	libgnuradio-fcdproplus.so.3.8.0 => /lib/arm-linux-gnueabihf/libgnuradio-fcdproplus.so.3.8.0 (0xb4f18000)
	librtlsdr.so.0 => /lib/arm-linux-gnueabihf/librtlsdr.so.0 (0xb4efa000)
	libgnuradio-uhd.so.3.8.2 => /lib/arm-linux-gnueabihf/libgnuradio-uhd.so.3.8.2 (0xb4ea2000)
	libuhd.so.3.15.0 => /lib/arm-linux-gnueabihf/libuhd.so.3.15.0 (0xb4500000)
	libhackrf.so.0 => /lib/arm-linux-gnueabihf/libhackrf.so.0 (0xb44ea000)
	libbladeRF.so.2 => /lib/arm-linux-gnueabihf/libbladeRF.so.2 (0xb445c000)
	libairspy.so.0 => /lib/arm-linux-gnueabihf/libairspy.so.0 (0xb4446000)
	libairspyhf.so.1 => /lib/arm-linux-gnueabihf/libairspyhf.so.1 (0xb4428000)
	libSoapySDR.so.0.7 => /lib/arm-linux-gnueabihf/libSoapySDR.so.0.7 (0xb43cf000)
	libfreesrp.so.0 => /lib/arm-linux-gnueabihf/libfreesrp.so.0 (0xb43b2000)
	libpulsecommon-14.2.so => /usr/lib/arm-linux-gnueabihf/pulseaudio/libpulsecommon-14.2.so (0xb432f000)
	libdbus-1.so.3 => /lib/arm-linux-gnueabihf/libdbus-1.so.3 (0xb42db000)
	libatomic.so.1 => /lib/arm-linux-gnueabihf/libatomic.so.1 (0xb42c2000)
	libGLESv2.so.2 => /lib/arm-linux-gnueabihf/libGLESv2.so.2 (0xb42a3000)
	libpng16.so.16 => /lib/arm-linux-gnueabihf/libpng16.so.16 (0xb4264000)
[... 45 lines cut ...]
	libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0xb109d000)
	libXau.so.6 => /lib/arm-linux-gnueabihf/libXau.so.6 (0xb108a000)
	libXdmcp.so.6 => /lib/arm-linux-gnueabihf/libXdmcp.so.6 (0xb1075000)
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0xb1044000)
	liblz4.so.1 => /lib/arm-linux-gnueabihf/liblz4.so.1 (0xb1017000)
	libgcrypt.so.20 => /lib/arm-linux-gnueabihf/libgcrypt.so.20 (0xb0f3a000)
	libnsl.so.2 => /lib/arm-linux-gnueabihf/libnsl.so.2 (0xb0f16000)
	libFLAC.so.8 => /lib/arm-linux-gnueabihf/libFLAC.so.8 (0xb0ed9000)
	libvorbis.so.0 => /lib/arm-linux-gnueabihf/libvorbis.so.0 (0xb0ea0000)
	libvorbisenc.so.2 => /lib/arm-linux-gnueabihf/libvorbisenc.so.2 (0xb0e0e000)
	libopus.so.0 => /lib/arm-linux-gnueabihf/libopus.so.0 (0xb0db0000)
	libogg.so.0 => /lib/arm-linux-gnueabihf/libogg.so.0 (0xb0d98000)
	libnsl.so.1 => /lib/arm-linux-gnueabihf/libnsl.so.1 (0xb0d73000)
	libbrotlidec.so.1 => /lib/arm-linux-gnueabihf/libbrotlidec.so.1 (0xb0d58000)
	libbsd.so.0 => /lib/arm-linux-gnueabihf/libbsd.so.0 (0xb0d36000)
	libgpg-error.so.0 => /lib/arm-linux-gnueabihf/libgpg-error.so.0 (0xb0d09000)
	libtirpc.so.3 => /lib/arm-linux-gnueabihf/libtirpc.so.3 (0xb0cd2000)
	libbrotlicommon.so.1 => /lib/arm-linux-gnueabihf/libbrotlicommon.so.1 (0xb0c9f000)
	libmd.so.0 => /lib/arm-linux-gnueabihf/libmd.so.0 (0xb0c84000)
```

When gqrx stops operation, the waterfall freezes, but the GUI can still be used. A new start of the DSP processing (first icon) does not start the operation again.
When gqrx is terminated and restarted, it works fine again.

[image]

closed 2022-01-07 by ernstz

vladisslav2011, 2022-01-07:
It looks like the device stops producing more samples.
You may try to re-select the device without stopping the DSP.

I suggest you to run gqrx from console - there may be libusb errors in the log.
```
aUcb transfer status: 1, canceling...
rtlsdr_read_async returned with -5
libusb: error [udev_hotplug_event] ignoring udev action bind
```

I had similar behavior due to buggy USB hub and loose connectors.
The USB hub sometimes drops some transfers when used with hackrf @20Msps. No problem at lower sample rates - runs well for several hours. When the drop occurs there is no libusb error, hackrf shows rx led lit, but it does not produce samples. 
Loose connectors/too long USB cable usually show errors like above.