gqrx-sdr/gqrx #1053: MSI SDR support?

evrial, 2022-01-10:
Tried to launch with sdrplay config but no success.
Overall the best SDR dongle in $50 range, msi001 msi2500 chip, just need support

vladisslav2011, 2022-01-10:
Thanks for a bug report.
Could you provide more information, please:
1. Gqrx version and installation type (built from source, Appimage, distribution packages)
2. GNU Radio version and installation type
3. gr-osmosdr version and installation type
4. Your operating system
5. Gqrx startup log when run from console

msi2500 chip should be supported by gr-osmosdr. Maybe you have to enable gqrx to access the usb device by adjusting udev settings...

douniwan5788, 2022-10-20:
> Thanks for a bug report. Could you provide more information, please:
>
> 1. Gqrx version and installation type (built from source, Appimage, distribution packages)
> 2. GNU Radio version and installation [...]

Mac OS release 2.15.9 does not include sdrplay support
>gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.3.0 
>built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya

vladisslav2011, 2022-10-20:
MIRISDR support was disabled in 'gr3.8' branch of gr-osmosdr by @Hoernchen commit 911082ff85dd3bdff986b8928b041644b3824e72 and then removed completely by @argilo commit fe03d837038884bfde2efe63cbf8ead4b846a3c8.
I've fixed some bugs both in libmirisdr and gr-osmosdr ('gr3.7' branch only, most recent changes are here https://github.com/vladisslav2011/gr-osmosdr/commits/ngrx ), but I have not rebased my changes onto master branch of gr-osmosdr. Fell free to cherry-pick...

Hoernchen, 2022-10-20:
The support that was disabled/removed refers to https://gitea.osmocom.org/sdr/libmirisdr and was originally introduced ten years ago as quick hack to make the arcane japanese isdb-t msi dongle work, but it appears that there are _other_ libmirisdrs now that support other, newer devices (?!)

This is basically the first time I'm hearing about this, so if this arcane, old mirisdr was indeed being used with much newer sdrs and newer libs with the same name I can just reintroduce it.

AsciiWolf, 2025-09-01:
It looks like that [libmirisdr-4](https://github.com/f4exb/libmirisdr-4) also supports original SDRplay devices. It would be great to have this support in Gqrx to avoid using the proprietary API and SoapySDRPlay3.