gqrx-sdr/gqrx #1073: macOS Monterey 12.1 UI Bugs

maxbundscherer, 2022-01-21:
Hi,

I am using qqrx 2.15.5 on macOS Monterey 12.1 UI. Unfortunately, there are some very annoying UI bugs - after updating to monterey. I have tried different gqrx version on different macs, but the issue remains.

The Pand.dB and Wf.dB slider in fft-settings tabs don't work correctly. Mostly the sliders set lower and upper levels to the same level and I cant move the upper and lower independent of each other. Only restarting gqrx helps and I can set the levels 2-3 times after having the same issue again.

Maybe some of u know what the problem is or can tell me where I can fix it.

Thank you!

[image]

maxbundscherer, 2022-01-22:
@LongnoseRob  - thank you for your answer. I have tried to click on the lock button, the bug remains (see attached screenshot). Yeah, it is a macOS only bug - on Linux, I have no problem.

[image]

argilo, 2022-01-22:
Thanks for the bug report. I was able to reproduce this in Gqrx 2.11.5 on macOS 12.1, so at least I know that it's not due to recent changes in Gqrx.

I also once on Linux observed the behaviour where the range extends all the way to the left even though no slider handle is there. I haven't been able to reproduce it since.

The ctkRangeSlider widget was copied into Gqrx back in 2016, and I suppose it could have bugs that came up since then.

maxbundscherer, 2022-01-23:
@argilo thank you for the information!

argilo, 2022-01-24:
It seems like there are bugs with the QSlider widget in Monterey: https://bugreports.qt.io/browse/QTBUG-98093

The ctkRangeSlider widget is based on QSlider, so it would probably inherit those bugs.

Gqrx is built with [Qt from Homebrew](https://formulae.brew.sh/formula/qt@5), which is presently at version 5.15.2 (released November 2020, well before Monterey). QSlider fixes may have been backported to later 5.15 releases, but I wouldn't be surprised if Homebrew never picks them up.

I suppose it may be necessary to update Gqrx to Qt 6 to get a version that properly supports Monterey. That would probably require some conditional compilation, since many of the Linux distributions that Gqrx runs on do not have Qt 6.

maxbundscherer, 2022-01-24:
@argilo - do you think it makes sense to update to qt6 in future gqrx versions? Then we should try this. Most likely we need to adjust something in the code.

argilo, 2022-01-24:
I knew that it would be necessary to upgrade to Qt 6 sooner or later, and the QSlider bugs are a good reason to prioritize that now. I'll make an attempt at it today and see how it goes.

maxbundscherer, 2022-01-24:
@argilo thank you very much!

argilo, 2022-01-26:
Gqrx works with Qt6! (See #1090 for details.)

I made a quick attempt to build a Mac version, but the packaging didn't work correctly. (A shared library was missing.) Hopefully it won't be hard to solve that problem.

maxbundscherer, 2022-01-26:
@argilo  - sounds great. Should I test it? Is the gt6-update on any branch?

argilo, 2022-01-26:
I've already merged #1090 into the master branch. If you're able to build from source on the Mac, install the `qt@6` package from homebrew, build Gqrx and take it for a spin.

When I added `qt@6` to the CI build (in https://github.com/gqrx-sdr/gqrx/commit/63634de340292ab55018a09c565f0dd3b59360e8), the packaged version was missing `libbrotlicommon.1.dylib`:
```
Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @loader_path/libbrotlicommon.1.dylib
Referenced from: /Users/argilo/Desktop/Gqrx.app/Contents/Frameworks/libbrotlidec.1.dylib
Reason: tried: '/Users/argilo/Desktop/Gqrx.app/Contents/Frameworks/libbrotlicommon.1.dylib' (no such file), '/usr/lib/libbrotlicommon.1.dylib' (no such file)Library not loaded: @rpath/QtDBus.framework/Versions/A/QtDBus
Referenced from: /Users/USER/Desktop/Gqrx.app/Contents/Frameworks/QtGui.framework/Versions/A/QtGui
Reason: tried: '/Users/argilo/Desktop/Gqrx.app/Contents/Frameworks/QtGui.framework/Versions/A/../../../QtDBus.framework/Versions/A/QtDBus' (no such file), '/Users/argilo/Desktop/Gqrx.app/Contents/MacOS/../libs/QtDBus.framework/Versions/A/QtDBus' (no such file), '/Users/argilo/Desktop/Gqrx.app/Contents/MacOS/../libs/QtDBus.framework/Versions/A/QtDBus' (no such file), '/Users/argilo/Desktop/Gqrx.app/Contents/MacOS/../libs/QtDBus.framework/Versions/A/QtDBus' (no such file), '/Users/argilo/Desktop/Gqrx.app/Contents/MacOS/../libs/QtDBus.framework/V
(terminated at launch; ignore backtrace)
```

argilo, 2022-01-27:
Success! I found an incantation that produces a working build. You can find a pre-release build here:

https://github.com/gqrx-sdr/gqrx/actions/runs/1757949976

It seems like the sliders are working better in this version. If you have a chance to try it, I'd appreciate your help with testing.

argilo, 2022-01-27:
I've since done some cleanup. Please use the build posted in #1094, which should be available shortly.

PR #1094 (2022-01-30): Build against Qt6 on macOS
Fixes #1073.

Now that Gqrx has been updated with Qt6 support (#1079, #1081, #1082, #1083, #1084, #1085, #1086, #1087, #1090), we should be able to switch over the macOS build.

For some reason, macdeployqt fails with a bunch of errors on the first run:
```
ERROR: Cannot resolve rpath "@rpath/QtGui.framework/Versions/A/QtGui"
ERROR:  using QSet("/Users/runner/work/gqrx/gqrx/lib")
ERROR: Cannot resolve rpath "@rpath/QtCore.framework/Versions/A/QtCore"
ERROR:  using QSet("/Users/runner/work/gqrx/gqrx/lib")
...
```
Running it a second time works around that problem.

Al [... cut]
closed 2022-01-30 by argilo