gqrx-sdr/gqrx #1075: BUG: flowgraph reconfiguration breaks audio recording

vladisslav2011, 2022-01-22:
When the flowgraph reconfiguration occurs (demodulator change, AM DC cancel change, global DC remove change, etc.), the audio recorder stops in background, leaving GUI button in pressed state.

Conditions:
gqrx built from git master, GNU Radio 3.7.11, Ubuntu 18.04

Steps to reproduce:
1. Start gqrx
2. Start the DSP
3. Tune to some frequency
4. Start audio recording
5. Toggle input controls -> DC remove
6. Wait some time
7. Stop audio recording
8. Check resulting audio file

Expected behavior:
The audio file contains all samples from recording start (4) to recording end (7).

Observed behavior:
The audio file contains only samples from recording start (4) to DC remove toggle (5). Samples from DC remove toggle (5) to recording end (7) are missing.

Primary cause: https://github.com/gnuradio/gnuradio/blob/main/gr-blocks/lib/wavfile_sink_impl.cc#L232-L241

Possible workarounds: none.

Possible fix: reimplement wavfile_sink to fix it locally, open PR to GNU Radio repository, wait for it to get merged, add conditional compilation directives...

argilo, 2022-01-23:
> Possible fix: reimplement wavfile_sink to fix it locally, open PR to GNU Radio repository, wait for it to get merged, add conditional compilation directives...

I'd suggest starting with a patch to GNU Radio. I don't want to maintain parallel block implementations in Gqrx.

vladisslav2011, 2022-01-26:
Another related issue: https://github.com/gqrx-sdr/gqrx/issues/400

> I'd suggest starting with a patch to GNU Radio.
I'll have to implement tag processing, so this will not be parallel implementation. A patch to GNU Radio...maybe later. After https://github.com/gqrx-sdr/gqrx/issues/109 and https://github.com/gqrx-sdr/gqrx/issues/946