gqrx-sdr/gqrx #1078: gqrx-2.15.4 segfaults if left idling for some time

cyberpunkrocker-zero, 2022-01-24:
Using gqrx-2.15.4 on Arch Linux, kernel 5.16.2-zen1-1-zen.

If I stop DSP processing and leave gqrx running on its own, and try to re-start DSP processing after some time (in this example arter an hour), gqrx segfaults and dumps a core. It does not happen immediately after stopping, but needs to be left running for some time. I don't know the shortest required time, but an hour is enough to crash it.

This did not happen on the previous Arch package gqrx-2.15.1, or earlier.

I didn't find any obvious error messages anywhere, except this from journalctl:
```
tammi 24 19:59:44 neverwinter audit[30254]: ANOM_ABEND auid=1000 uid=1000 gid=1000 ses=3 subj==unconfined pid=30254 comm="gqrx" exe="/usr/bin/gqrx" sig=11 res=1
tammi 24 19:59:44 neverwinter kernel: gqrx[30254]: segfault at 7f3185a48be8 ip 00007f2d8e66f0bc sp 00007ffd51b003a8 error 4 in orcexec.WyEOma (deleted)[7f2d8e66f000+10000]
tammi 24 19:59:44 neverwinter kernel: Code: 48 8d 40 08 48 8d 52 08 48 8d 76 04 83 7f 10 00 74 50 44 8b 47 10 90 90 90 90 90 90 90 90 f3 0f 7e 06 f3 0f 7e 0e 66 0f 62 c1 <f3> 0f 6f 0a 0f 59 c8 66 0f 7f 08 f3 0f 7e 46 08 f3 0f 7e 4e 08 66
tammi 24 19:59:44 neverwinter kernel: audit: type=1701 audit(1643047184.432:275): auid=1000 uid=1000 gid=1000 ses=3 subj==unconfined pid=30254 comm="gqrx" exe="/usr/bin/gqrx" sig=11 res=1
tammi 24 19:59:44 neverwinter audit: BPF prog-id=50 op=LOAD
tammi 24 19:59:44 neverwinter audit: BPF prog-id=51 op=LOAD
tammi 24 19:59:44 neverwinter audit: BPF prog-id=52 op=LOAD
tammi 24 19:59:44 neverwinter kernel: audit: type=1334 audit(1643047184.446:276): prog-id=50 op=LOAD
tammi 24 19:59:44 neverwinter kernel: audit: type=1334 audit(1643047184.446:277): prog-id=51 op=LOAD
tammi 24 19:59:44 neverwinter kernel: audit: type=1334 audit(1643047184.446:278): prog-id=52 op=LOAD
tammi 24 19:59:44 neverwinter systemd[1]: Started Process Core Dump (PID 33306/UID 0).
tammi 24 19:59:44 neverwinter audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-coredump@3-33306-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
tammi 24 19:59:44 neverwinter kernel: audit: type=1130 audit(1643047184.447:279): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-coredump@3-33306-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
tammi 24 19:59:44 neverwinter systemd-coredump[33307]: Resource limits disable core dumping for process 30254 (gqrx).
tammi 24 19:59:44 neverwinter systemd-coredump[33307]: [🡕] Process 30254 (gqrx) of user 1000 dumped core.
tammi 24 19:59:44 neverwinter systemd[1]: systemd-coredump@3-33306-0.service: Deactivated successfully.
tammi 24 19:59:44 neverwinter kernel: audit: type=1131 audit(1643047184.476:280): pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-coredump@3-33306-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
tammi 24 19:59:44 neverwinter audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj==unconfined msg='unit=systemd-coredump@3-33306-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
tammi 24 19:59:44 neverwinter audit: BPF prog-id=0 op=UNLOAD
tammi 24 19:59:44 neverwinter audit: BPF prog-id=0 op=UNLOAD
tammi 24 19:59:44 neverwinter audit: BPF prog-id=0 op=UNLOAD
tammi 24 19:59:44 neverwinter kernel: audit: type=1334 audit(1643047184.526:281): prog-id=0 op=UNLOAD
tammi 24 19:59:44 neverwinter kernel: audit: type=1334 audit(1643047184.526:282): prog-id=0 op=UNLOAD
tammi 24 19:59:44 neverwinter kernel: audit: type=1334 audit(1643047184.526:283): prog-id=0 op=UNLOAD
```

argilo, 2022-01-24:
Thanks for the bug report!

It looks like the problem is an integer overflow here:

https://github.com/gqrx-sdr/gqrx/blob/60f6910bf65bad6129ef41af440dbceacd3b93dc/src/dsp/rx_fft.cpp#L125

The same overflow can also happen here:

https://github.com/gqrx-sdr/gqrx/blob/60f6910bf65bad6129ef41af440dbceacd3b93dc/src/dsp/rx_fft.cpp#L324

The underlying bug has probably been around for a while, but I #1051 made it worse because the change from `unsigned int` to `int` means that the overflow can now result in `(int)(diff.count() * d_quadrate * 1.001)` being negative.

I'll add bounds checks to prevent the overflows.

argilo, 2022-01-24:
#1080 should fix this.

PR #1080 (2022-01-24): Add bounds checks to prevent integer overflows
Fixes #1078.

Integer overflows can happen in these two places, if the DSP is paused for long enough:

https://github.com/gqrx-sdr/gqrx/blob/60f6910bf65bad6129ef41af440dbceacd3b93dc/src/dsp/rx_fft.cpp#L125

https://github.com/gqrx-sdr/gqrx/blob/60f6910bf65bad6129ef41af440dbceacd3b93dc/src/dsp/rx_fft.cpp#L324

To fix this, I've added a bounds check to `diff` to ensure that it can't be longer than the duration necessary to fill the entire circular buffer.
closed 2022-01-24 by argilo