gqrx-sdr/gqrx #1098: Loop IQ playback

MightyPork, 2022-01-29:
I often find it annoying that the IQ playback tool resets my cursor position and zoom when it ends (and I restart playback), particularly when I'm trying to decode CW signals from a shorter sample.

I'm not familiar with C++ or Qt, now I'm working on a patch to do it, but so far I only managed to add looping, it still resets the zoom.

```patch
diff --git a/src/qtgui/iq_tool.cpp b/src/qtgui/iq_tool.cpp
--- a/src/qtgui/iq_tool.cpp	(revision a7912a786114fe368b8c5370cb53133d036c01e4)
+++ b/src/qtgui/iq_tool.cpp	(date 1643480384425)
@@ -292,6 +292,13 @@
             ui->slider->setValue(val+1);
             ui->slider->blockSignals(false);
             refreshTimeWidgets();
+        } else {
+            // restart from the beginning
+            on_playButton_clicked(true);
+            ui->slider->blockSignals(true);
+            ui->slider->setValue(0);
+            ui->slider->blockSignals(false);
+            refreshTimeWidgets();
         }
     }
     if (is_recording)
@@ -377,3 +384,7 @@
     if (center_ok)
         center_freq = center;
 }
```

MightyPork, 2022-01-29:
Hmm, the patch is actually working less reliably than I thought. I must have missed some part of the puzzle, the playback still stops after the full run time. It seemingly works when I skip ahead using the slider, but not when I let it run at its own pace ...

vladisslav2011, 2022-01-31:
I have already implemented this feature in https://github.com/gqrx-sdr/gqrx/pull/1016, but this PR needs a rebase on top of current master, compilation fixes (does not build with GNU Radio head), and should be split into multiple PR's (loop playback, buffered reader, buffered writer, switch to GNU Radio buffers).

vladisslav2011, 2022-02-18:
I've separated this feature into a different PR (on top of iq_tool_formats): #1108. You may give it a try and make a review.
The slider position gets stuck at the end as it is not possible to track actual file position with current GNU Radio file_source implementation. Try #1016 if you want accurate slider position (I'll rebase it on top of current master soon)