gqrx-sdr/gqrx #1099: Error to install gqrx

jp-polito, 2022-01-29:
Hi! I'm trying install gqrx in my notebook, but after using command $ cmake.. this error appears, Any suggestions for me to solve this problem?

CMake Error at CMakeLists.txt:132 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency spdlog could not be found.

argilo, 2022-01-30:
As noted in the error message, you need to install spdlog. If you're on a Debian-based operating system, you could do that with `sudo apt install libspdlog-dev`.

closed 2022-01-30 by argilo

jp-polito, 2022-01-30:
Thanks for answer! I am using the ubuntu 20.04 after install the libspdlog-dev I tried running cmake again, and it got this error.
CMake Error at src/CMakeLists.txt:71 (add_executable):
  Target "gqrx" links to target "Portaudio::Portaudio" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

argilo, 2022-01-30:
`sudo apt install portaudio19-dev`