gqrx-sdr/gqrx #1101: 2.15.8 Segmentation fault once

idallen, 2022-02-03:
**Executive summary (details follow)**

- I compiled 2.15.8 on my Ubuntu 20.04 system.
    - Linux linux 5.13.0-27-generic 29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
- I got a segmentation fault and core dump the first time I ran 2.15.8.
    -  6405539 Feb  3 10:55 _usr_bin_gqrx.1000.crash
- I ran the old version 2.14.4 and it worked.
- I ran 2.15.8 under gdb and it worked.
- I ran 2.15.8 again and it worked, and continues to work.

**Details**

    $ gqrx --help
    Gqrx software defined radio receiver 2.15.8

    $ gqrx
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    gr::log :WARN: file_source0 - file size is not a multiple of item size
    Resampling audio 96000 -> 48000
    BandPlanFile is /home/idallen/.config/gqrx/bandplan.csv
    BookmarksFile is /home/idallen/.config/gqrx/bookmarks.csv
    [INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5

    RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

    libusb: debug [libusb_get_device_descriptor]
    Segmentation fault (core dumped)
    Exit 139

    $ ls -l /var/crash/*gqrx*
    -rw-r----- 1 idallen whoopsie 6405539 Feb  3 10:55 /var/crash/_usr_bin_gqrx.1000.crash

    $ gqrx.old --help
    Gqrx software defined radio receiver v2.14.4-21-g706a72d

    $ gqrx.old
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    gr::log :WARN: file_source0 - file size is not a multiple of item size
    Resampling audio 96000 -> 48000
    BandPlanFile is /home/idallen/.config/gqrx/bandplan.csv
    BookmarksFile is /home/idallen/.config/gqrx/bookmarks.csv
    [INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5

    RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

    Detached kernel driver
    Found Rafael Micro R820T tuner
    Reattached kernel driver
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    Using device #0 Realtek RTL2838UHIDIR SN: 00000001
    Detached kernel driver
    Found Rafael Micro R820T tuner
    [R82XX] PLL not locked!
    Enabled direct sampling mode, input 2
    Reattached kernel driver

    $ gdb gqrx
    +(gdb) run
    ... works fine; no segmentation fault ...

    $ gqrx
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    gr::log :WARN: file_source0 - file size is not a multiple of item size
    Resampling audio 96000 -> 48000
    BandPlanFile is /home/idallen/.config/gqrx/bandplan.csv
    BookmarksFile is /home/idallen/.config/gqrx/bookmarks.csv
    [INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5

    RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

    Detached kernel driver
    Found Rafael Micro R820T tuner
    Reattached kernel driver
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    gr::log :WARN: file_source1 - file size is not a multiple of item size
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    Using device #0 Realtek RTL2838UHIDIR SN: 00000001
    Detached kernel driver
    Found Rafael Micro R820T tuner
    [R82XX] PLL not locked!
    Enabled direct sampling mode, input 2
    Allocating 15 zero-copy buffers
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    gr::log :WARN: file_source0 - file size is not a multiple of item size
    Reattached kernel driver
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    Using device #0 Realtek RTL2838UHIDIR SN: 00000001
    Detached kernel driver
    Found Rafael Micro R820T tuner
    [R82XX] PLL not locked!
    Enabled direct sampling mode, input 2
    Allocating 15 zero-copy buffers
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    gr::log :WARN: file_source0 - file size is not a multiple of item size
    Reattached kernel driver
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    Using device #0 Realtek RTL2838UHIDIR SN: 00000001
    Detached kernel driver
    Found Rafael Micro R820T tuner
    [R82XX] PLL not locked!
    Enabled direct sampling mode, input 2
    Allocating 15 zero-copy buffers

    RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (hw:0,0), Device or resource busy.

    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    gr::log :WARN: file_source0 - file size is not a multiple of item size
    Reattached kernel driver
    gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
    Using device #0 Realtek RTL2838UHIDIR SN: 00000001
    Detached kernel driver
    Found Rafael Micro R820T tuner
    [R82XX] PLL not locked!
    Enabled direct sampling mode, input 2
    Allocating 15 zero-copy buffers
    Starting UDP streaming, Host: localhost, Port: 7355, Mono
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Disconnected UDP streaming
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Resampling audio 96000 -> 48000
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Allocating 15 zero-copy buffers
    Reattached kernel driver

... and it continues to work fine, no segmentation faults ...

argilo, 2022-02-03:
Thanks for the report.

Are you able to share the crash file and the corresponding gqrx binary? In the absence of a reliable reproduction steps, those are just about the only clues available.

I also note that `libusb: debug [libusb_get_device_descriptor]` appears in the crash scenario but not the others. This function is called by librtlsdr, so it could be that the issue lies there.

idallen, 2022-02-03:
[bad.tar.gz](https://github.com/gqrx-sdr/gqrx/files/7997453/bad.tar.gz)

`tar` file contains: `gqrx` binary, crash file, entire build directory (including same binary `gqrx-2.15.8/build/src/gqrx`)

argilo, 2022-02-03:
The stack trace shows that the crash was indeed inside `libusb_get_device_descriptor`, but that it was called by libuhd rather than librtlsdr:
```
#0  0x00007f1e0c0008d0 in ?? ()
#1  0x00007f1e4405e47a in ?? () from /usr/local/lib/x86_64-linux-gnu/libusb-1.0.so.0
#2  0x00007f1e4405e60e in ?? () from /usr/local/lib/x86_64-linux-gnu/libusb-1.0.so.0
#3  0x00007f1e4406119c in libusb_get_device_descriptor () from /usr/local/lib/x86_64-linux-gnu/libusb-1.0.so.0
#4  0x00007f1e45e3dff6 in ?? () from /usr/local/lib/x86_64-linux-gnu/libuhd.so.3.15.0
#5  0x00007f1e45e3bdc0 in ?? () from /usr/local/lib/x86_64-linux-gnu/libuhd.so.3.15.0
#6  0x00007f1e45e3c774 in uhd::transport::usb_device_handle::get_device_list(std::vector<std::pair<unsigned short, unsigned short>, std::allocator<std::pair<unsigned short, unsigned short> > > const&) () from /usr/local/lib/x86_64-linux-gnu/libuhd.so.3.15.0
#7  0x00007f1e45e3cc9b in uhd::transport::usb_device_handle::get_device_list(unsigned short, unsigned short) ()
   from /usr/local/lib/x86_64-linux-gnu/libuhd.so.3.15.0
#8  0x00007f1e45c3c66f in ?? () from /usr/local/lib/x86_64-linux-gnu/libuhd.so.3.15.0
#9  0x00007f1e45be8c66 in ?? () from /usr/local/lib/x86_64-linux-gnu/libuhd.so.3.15.0
#10 0x00007f1e45e872be in ?? () from /usr/local/lib/x86_64-linux-gnu/libuhd.so.3.15.0
#11 0x00007f1e457d09ad in ?? () from /usr/local/lib/x86_64-linux-gnu/libuhd.so.3.15.0
#12 0x00007f1e46a144df in futex_wake (private=0, processes_to_wake=2147483647, futex_word=0x7f1e0c0023c0)
    at ../sysdeps/nptl/futex-internal.h:364
#13 clear_once_control (arg=0x7f1e0c0023c0) at pthread_once.c:38
#14 0x00007f1e46a14530 in __GI___pthread_once (once_control=0x7f1e0c0023c0, init_routine=0x4) at pthread_once.c:140
#15 0x00007f1de4001198 in ?? ()
#16 0x0000000000000000 in ?? ()
```
Gqrx does not appear in the stack trace, so this is probably some libuhd initialization code. I suspect that the bug lies outside Gqrx.

argilo, 2022-02-09:
I'll close this off for now, since I'm unable to reproduce and there's not a clear indication that the bug lies in Gqrx. If you're able to get more information about this, let me know.

closed 2022-02-09 by argilo