gqrx-sdr/gqrx #1103: My device disappeared when I open Gqrx

langonginc, 2022-02-09:
I bought an SDR device (RTL-SDR), and I connected and listened successfully yesterday. But today, when I open GQRX, It seemed the application cannot find my device (I connected with USB, and the light is on normally). How can I fix it?

P.S. After that, I restarted system and used `rtl_test` found my device.

`Archlinux x64 5.15.21-1-lts` with `gqrx 2.15.8-1` and `gnuradio 3.9.4.0-2` installed by `pacman`.

Thank you!

argilo, 2022-02-09:
I don't really have enough information to diagnose the problem here. The only thing that comes to mind is that perhaps another application was using the device.

Does unplugging the device and plugging it back in help?

langonginc, 2022-02-10:
> Does unplugging the device and plugging it back in help?

I tried, but it didn't work.

langonginc, 2022-03-25:
Thanks for help, I'm now fixed it. My device was broken. The seller exchanged of it and now can work.

closed 2022-03-25 by langonginc