gqrx-sdr/gqrx #1104: Apple Silicon / M1 builds?

drwetter, 2022-02-09:
Are there any plans to supply builds for the new architecture? Even hombrew fails (_Bad CPU type in executable_).

Is there any obstacles since qt6 works on aarch64?

argilo, 2022-02-09:
I would like the macOS builds to have M1 support. I just don't have the expertise (or hardware) to work on this at the moment.

One obstacle is that the Gqrx build currently uses gnuradio from Homebrew, which does not support M1:

https://github.com/gqrx-sdr/gqrx/blob/f108a292e400ada937cad27e80748eca23da448f/.github/workflows/build.yml#L89

https://formulae.brew.sh/formula/gnuradio

One possibility would be to build Gqrx's dependencies from source, instead of installing them from Homebrew. This would slow down builds, but perhaps they could be sped up again using ccache.

If anyone has expertise in building software with M1 support, I'd appreciate help.

drwetter, 2022-02-10:
FYI (but probably doesn't help much): I tried yesterday to compile gnuradio via homebrew. It stopped when compiling volk as dependency (error when compiling volk_machine_neonv8.c. It did only state the clang command but not the actual error).

vielmetti, 2022-04-04:
The "volk doesn't build on M1" issue is https://github.com/gnuradio/volk/issues/428 upstream, which is said to be fixed by https://github.com/google/cpu_features/commit/69d39934e83be9a0133d2850f15f5debd5eba06f reported in https://github.com/google/cpu_features/issues/179 with a PR in https://github.com/google/cpu_features/pull/209 - which as much to say that it might be worth revisiting this issue if indeed the problems are resolved.

vielmetti, 2022-04-04:
I was just now able to install "volk" with Home-brew, getting this:

==> Pouring volk--2.5.1_1.arm64_monterey.bottle.tar.gz

and no errors.

vielmetti, 2022-04-04:
Yup, and GNU Radio now installs OK

==> Pouring gnuradio--3.9.3.0_4.arm64_monterey.bottle.tar.gz

dsegel, 2022-04-16:
I just installed v2.15.9 (I don't understand the version numbering) via homebrew (brew install gqrx) and it runs fine on my M1 MacBook Air.

`==> Downloading https://github.com/gqrx-sdr/gqrx/releases/download/v2.15.9/Gqrx-2.15.9.dmg
==> Downloading from https://objects.githubusercontent.com/github-production-release-asset-2e65be/1033271/0074ad10-fcdb-4e9d-abac-c837c0
######################################################################## 100.0%
==> Installing Cask gqrx
==> Moving App 'Gqrx.app' to '/Applications/Gqrx.app'
==> Linking Binary 'gqrx.wrapper.sh' to '/opt/homebrew/bin/gqrx'
🍺  gqrx was successfully installed!
`

Forst, 2022-07-18:
I gave a quick and dirty go at building Gqrx for M1. Looking at `.github/workflow/build.yml`, the only major difference is Homebrew is installed in `/opt/homebrew` on Apple Silicon platforms instead of `/usr/local` on Intel. This affects all mentions of `/usr/local` in the build scripts, and needs to be defined as the cmake install prefix for certain dependencies.

Built and tested on macOS 12.4. Tested with RTL-SDR (USB), seems to work alright:

[image]

[image]

I put together a build script from `.github/workflow/build.yml` and `macos_bundle.sh`, posted here: https://gist.github.com/Forst/db68106136be3380086e3c38be094d99

Note that I am a total noob when it comes to building apps for macOS, so I just changed things around until it worked :)

danielktdoranie, 2023-03-03:
> I gave a quick and dirty go at building Gqrx for M1. Looking at `.github/workflow/build.yml`, the only major difference is Homebrew is installed in `/opt/homebrew` on Apple Silicon platforms instead [...]

Hey sir any hope of you dropping that .dmg file somewhere so I can try it?

Forst, 2023-03-10:
> Hey @Forst any hope of you dropping that .dmg file somewhere so I can try it?

@danielktdoranie Here you go: https://drive.proton.me/urls/N8C7JTHZB0#HJbeqbOBA2HA. It's the same dmg from July last year, luckily I had it lying around.

Be wary of running programs from strangers on the internet. I posted the build script in my earlier comment so you could reproduce the results yourself with code you can audit/verify.

danielktdoranie, 2023-03-11:
> @danielktdoranie Here you go: https://drive.proton.me/urls/N8C7JTHZB0#HJbeqbOBA2HA. It's the same dmg from July last year, luckily I had it lying around.
>
> Be wary of running programs from strangers [...]

Thanks. Yeah, I need to learn to compile programs myself but I really wish the developer would put this version up on the main webpage for SDR normies like myself. Thanks for providing this!

argilo, 2024-04-05:
I see conda-forge has an osx-arm64 build: https://anaconda.org/conda-forge/gqrx

Gqrx Linux builds were recently switched over to conda (#1301), so perhaps conda could be a replacement for homebrew on macOS as well.

kgarrels, 2024-04-05:
I am building gqrx with radioconda on macOS 12, works without problem.
I am still using Intel, and I am using local builds, not github actions.

kgarrels, 2024-10-18:
in the meantime, I switched to ARM, gqrx build and works fine with radioconda.
I use Qt5, Qt6 is not yet avaiable for mac on radioconda.

Can I help somhow? I am not experienced with github actions, but I can record all steps from a clean macos 14 install until build. Would that help?

yuzawa-san, 2024-10-21:
i gave my attempt at this in https://github.com/gqrx-sdr/gqrx/pull/1388 please try out the build artifacts. looks like they are still not signed in the PR build probably for security reasons. maybe the maintainers need to trigger a rerun so the secrets will populate. my friend had trouble opening the unsigned app in their arm64 sequoia environment.

PR #1388 (2024-11-09): Build both x86_64 and arm64 macOS DMG's
fixes #1104

- i tried doing the conda version, but found that the qt in there does not bundle properly and was bundling the libraries twice. i may dig into it further later, but this still uses the brew installation methodology
- there are a few gotchas: 
- the macos-13 GH runner is x86_64 only in the free tier and the macos-14 GH runner is arm64 only in the free tier, so technically the two produced artifacts have different OS baselines. both brew and conda do not produce universal2 binaries and seem to pin to the OS baselines, so not much i think we c [... cut]
closed 2024-11-09 by argilo