gqrx-sdr/gqrx #1109: gqrx crashes on similar system

8fXvSYUcRD7u, 2022-03-10:
Hello, I have one PC and one Laptop, which both run OpenSUSE Tumbleweed, and are pretty similar from a software side.
I initially ran sdrpp, which worked without problems with my RTL-SDR on both systems, but later I switched to gqrx, which runs just fine on my PC, but crashes on my laptop.

I tried the AppImage, which, again, runs fine on my PC but crashes on my Laptop.

Gqrx version (on both systems): `2.15.8-1.6`
Gqrx appimage: `2.15.8`
uname (on both systems): `Linux 5.14.14-2-default #1 SMP Thu Oct 21 05:05:03 UTC 2021 (2b5383f) x86_64 x86_64 x86_64 GNU/Linux`

gqrx run from Terminal on Laptop:(**CRASHES**)
```

┬─[user@system][/home/user]─[21:04:52]
╰─>$ gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.5.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx
Resampling audio 96000 -> 48000
terminate reached from thread id: 7f024638c0c0Got std::runtime_error
send_to: Das Netzwerk ist nicht erreichbar [system:101]
0x7f024970b6be: (gr::tagged_stream_block::check_topology(int, int)+0x44e)
0x7f02491e119c: (std::rethrow_exception(std::__exception_ptr::exception_ptr)+0x7c)
0x7f02491e01f9: (__cxa_free_dependent_exception+0x79)
0x7f02491e0927: (__gxx_personality_v0+0x87)
0x7f024903b484: (_Unwind_GetTextRelBase+0x1ee4)
0x7f024903bebe: (_Unwind_Resume+0x12e)
0x7f024984516c: -- error: unable to obtain symbol name for this frame
0x7f02498e5cb5: (gr::blocks::uchar_to_float::make()+0x655)
0x7f02498e5d33: (gr::blocks::uchar_to_float::make()+0x6d3)
0x563f4b5ee0da: (receiver::escape_filename(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)+0xbaa)
0x563f4b5b12a5: -- error: unable to obtain symbol name for this frame
0x563f4b60f17f: (make_udp_sink_f()+0x2f)
0x563f4b5ebb7a: (receiver::receiver(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, unsigned int)+0x8ea)
0x563f4b5da25b: (MainWindow::MainWindow(QString const&, bool, QWidget*)+0x36b)
0x563f4b5bff88: (main+0x13b8)
0x7f0248e3b630: (__libc_start_call_main+0x82)
0x7f0248e3b6f0: (__libc_start_main+0x82)
0x563f4b5c5a55: (_start+0x25)
fish: Job 1, 'gqrx' terminated by signal SIGABRT (Abbruch)

```
gqrx run from terminal on PC: (**WORKS**)

```
┬─[net@localhost][/home/net]─[21:04:03]
╰─>$ gqrx
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.5.0
     built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx
     gr::log :WARN: file_source0 - file size is not a multiple of item size
     Resampling audio 96000 -> 48000
     gr::log :ERROR: fft::import_wisdom - can't import wisdom from /home/net/.gr_fftw_wisdom
     BandPlanFile is /home/net/.config/gqrx/bandplan.csv
     BookmarksFile is /home/net/.config/gqrx/bookmarks.csv
     [INFO] [UHD] linux; GNU C++ version 11.2.1 20220103 [revision d4a1d3c4b377f1d4acb34fe1b55b5088a3f293f6]; Boost_107800; UHD_4.1.0.5-0-unknown
     Found Rafael Micro R820T tuner
     gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.5.0
     built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx
     Using device #0 Realtek RTL2838UHIDIR SN: 00000001
     Found Rafael Micro R820T tuner
     [R82XX] PLL not locked!
     [R82XX] PLL not locked!
     [R82XX] PLL not locked!

```

gqrx Appimage run from terminal on Laptop: (**CRASHES**)

```
┬─[user@system][/home/user]─[21:05:22]
╰─>$ ./Gqrx-2.15.8-x86_64.AppImage
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Resampling audio 96000 -> 48000
gr::fft: can't import wisdom from /home/user/.gr_fftw_wisdom
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  send_to: Das Netzwerk ist nicht erreichbar
fish: Job 1, './Gqrx-2.15.8-x86_64.AppImage' terminated by signal SIGABRT (Abbruch)

```
gqrx Appimage run from terminal on PC: (**WORKS**)

```
┬─[net@localhost][/home/net]─[21:05:30]
╰─>$ ./Gqrx-2.15.8-x86_64.AppImage
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

     gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
     built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
     gr::log :WARN: file_source0 - file size is not a multiple of item size
     Resampling audio 96000 -> 48000
     gr::fft: can't import wisdom from /home/net/.gr_fftw_wisdom
     BandPlanFile is /home/net/.config/gqrx/bandplan.csv
     BookmarksFile is /home/net/.config/gqrx/bookmarks.csv
     [ERROR] SoapySSDPEndpoint::sendTo(udp://[ff02::c]:1900) = -1
       sendto(udp://[ff02::c]:1900) [99: Die angeforderte Adresse kann nicht zugewiesen werden]

```

If you need any more information, let me know, I would be really thankful if you could help me.

vladisslav2011, 2022-03-11:
Hello.

>send_to: Das Netzwerk ist nicht erreichbar [system:101]

It looks, like you have not enabled ipv6 support on your laptop or you have network configuration problem.
1. You may try this PR https://github.com/gqrx-sdr/gqrx/pull/1065 to check for this issue to be fixed (gqrx should start and work correctly, but enabling UDP streaming should crash it) if you can build gqrx from source.
2. You may try to edit ~/.config/gqrx/default.conf: open the file with text editor, find "[audio]" section, set "udp_host" option to some valid host. "127.0.0.1" should work.

8fXvSYUcRD7u, 2022-03-11:
As gqrx doesn't even create its config, I manually added a default.conf with 
```
[audio]
udp_host=127.0.0.1

```
Which didn't change anything
I also tried copying the config from my PC, with no success.

As you mentioned, I looked at  https://github.com/gqrx-sdr/gqrx/pull/1065#issue-1104209540 and the corresponding issue: https://github.com/gqrx-sdr/gqrx/issues/467#issue-196283889

while commenting out `127.0.0.1 localhost` in `/etc/hosts` didn't change anything,
but removing `localhost` from
```
::1             localhost ipv6-localhost ipv6-loopback
```
lead to gqrx working like expected!
For me that seems to be good enough, that I am leaving it like this, but maybe I'll try to compile it one day.
For now, I'll close this issue.

Thanks!

closed 2022-03-11 by 8fXvSYUcRD7u

rag2, 2022-03-11:
Localhost is generally used for an IPv4 loopback route. If you kept the 
localhost definition commented out, then probably it defaulted to the 
IPv4 address 127.0.0.1. Setting it to an IPv6 address confused GQRX.

On 11/03/2022 10:44, 8fXvSYUcRD7u wrote:
>
> but removing |localhost| from
>
> |::1 localhost ipv6-localhost ipv6-loopback |
>
> lead to gqrx working like expected!
>

dranch, 2022-03-11:
Most modern Linux machines have some form of an IPv6 localhost defined in their /etc/hosts file.  An Ubuntu 20.04 host has the following but the IPv6 one isn't called "localhost" but "ipv6-localhost"
--
127.0.0.1       localhost
::1                  ip6-localhost ip6-loopback
--