gqrx-sdr/gqrx #1111: GQRX start takes long

Ta-2om, 2022-03-11:
Hello,

is not a big deal, but new installed (from the source) GQRX is taking too long to start.
I can live with that.
Here is report when running from console. Any suggestions? Yes I played too much around with other apps and their dependencies.

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace soapy redpitaya 
file_source :warning: file size is not a multiple of item size
Resampling audio 96000 -> 48000
BandPlanFile is /home/ersoy/.config/gqrx/bandplan.csv
BookmarksFile is /home/ersoy/.config/gqrx/bookmarks.csv
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.1.1
built-in source types: file rtl rtl_tcp uhd hackrf bladerf rfspace soapy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000110
Found Rafael Micro R820T/2 tuner

argilo, 2022-03-12:
How long exactly does it take (in seconds)?

Could you run Gqrx in gdb, press Ctrl+C during a slow startup, and print out the backtrace with `bt`? That will help determine what is slowing things down. Doing this multiple times will help confirm the source of the issue.

vladisslav2011, 2022-03-15:
I'm experiencing this issue too. Gqrx starts in about 4-5 seconds on my PC. Many gr-osmosdr drivers enumerate all usb devices trying to find something compatible. This takes some time.
Gqrx starts in 1-2 seconds when there are no local SDR devices connected (on the laptop with rtl-tcp source).

This block of code https://github.com/gqrx-sdr/gqrx/blob/master/src/qtgui/ioconfig.cpp#L143-L161 executes in about 4 seconds on my PC. Commenting out actual device enumeration on line https://github.com/gqrx-sdr/gqrx/blob/master/src/qtgui/ioconfig.cpp#L143 ( `osmosdr::devices_t devs/* = osmosdr::device::find()*/;` ) makes gqrx start in 1-2 seconds. But this makes device list empty...