gqrx-sdr/gqrx #1112: Crashing on start after SDRplay support implemented in gr-osmosdr with flag -DENABLE_NONFREE=TRUE

A2Sumie, 2022-03-14:
Self compiled.
Platform is Mac. Happens without plugging my rsp in.
Error goes like

# XXX @ XXXMacBook-Pro in ~/gadgets/gqrx.git/build on git:master o [18:31:18]
$ gqrx
dyld[44158]: symbol not found in flat namespace '_mir_sdr_Init'
[1]    44158 abort      gqrx

without the flag, at least gqrx could launch.

rag2, 2022-03-14:
It makes *far* more sense to start GQRX with the *-e* flag if the 
previous SDR hardware is no longer present. Why? Because:

    1) the problem is most probably in the SDR hardware libraries (SOAPY
    &c.)

    2) though that part of GQRX could probably have been defensively
    programmed, it wasn't. But now it is a mature piece of code, and
    attempting to undertake a *major* restructuring is likely to
    introduce more bugs than it fixes.

    3) there is already a sane workaround (restart GQRX and wait for the
    *Configure I/O devices* dialog box.)

HTH, 73,

Robin, G8DQX

PS: GQRX is now over a decade old. Though some new functions are being 
added, it is essentially on a care & maintenance régime. [The -e flag 
throws up the *Configure I/O devices* dialog box at GQRX start, btw.]

On 14/03/2022 10:50, A2Sumie wrote:
>
> Self compiled.
> Platform is Mac. Happens without plugging my rsp in.
> Error goes like
>
>
>   XXX @ XXXMacBook-Pro in ~/gadgets/gqrx.git/build on git:master o
>   [18:31:18]
>
> $ gqrx
> dyld[44158]: symbol not found in flat namespace '_mir_sdr_Init'
> [1] 44158 abort gqrx
>
> without the flag, at least gqrx could launch.
>
> Message ID: ***@***.***>

A2Sumie, 2022-03-15:
Problem persists with flag "-e". Crashes instantly and same error outputs.

> It makes *far* more sense to start GQRX with the *-e* flag if the previous SDR hardware is no longer present. Why? Because: 1) the problem is most probably in the SDR hardware libraries (SOAPY &c.) [...]

vladisslav2011, 2022-03-15:
It looks like libsdrplay_api.so is compiled with `libc++` and you are compiling gr-osmosdr with `libstdc++`. Try compiling gr-osmosdr with `libc++`.

A2Sumie, 2022-03-16:
> It looks like libsdrplay_api.so is compiled with `libc++` and you are compiling gr-osmosdr with `libstdc++`. Try compiling gr-osmosdr with `libc++`.

In CmakeLists.txt of gr-osmosdr  I added lines 
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -stdlib=libc++")
set(CMAKE_CXX_COMPILER "clang++")
set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -stdlib=libc++ -lc++abi")`
before # Compiler specific setup and
`
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -stdlib=libc++")
set(CMAKE_EXE_LINKER_FLAGS "${CMAKE_EXE_LINKER_FLAGS} -stdlib=libc++ -lc++abi")`
after #misc in # Compiler specific setup

Same errors after recompiling and reinstalling both, have i missed something?

vladisslav2011, 2022-03-22:
There is no sdrplay legacy API for MacOS ARM, so you may either update gr-osmosdr to support sdrplay API v3 or build gr-osmosdr with soapysdr support and build https://github.com/pothosware/SoapySDRPlay3 or switch to using libmirisdr-4 (this would require some reverse engineering, some bug fixing and maybe usage of nonfree firmware blob. You may pick some cherry from my forks.)