gqrx-sdr/gqrx #1114: Failed to open rtlsdr device

dltkddnr04, 2022-03-21:
Hello.
When I finish setting up the device and click the OK button, the following warning window appears.
And I can't use SDR through the program.

[image]
[image]

Does anyone know a solution to this problem?
I think it could be a problem with macos beta.
thanks.

Here are the specs I use
- Macbook Pro 13' 2015
- macOS 12.3 (21E230)
- GQRX 2.15.8
- RTL_SDR dongle(rtl2838u chip, blue one)

vladisslav2011, 2022-03-22:
Hello.
Try running gqrx with sudo.
If it helps, then read here: https://developer.apple.com/documentation/bundleresources/entitlements/com_apple_vm_device-access

dltkddnr04, 2022-03-26:
I tried with sudo and this is the same as before.
I can't understand what happened.

vladisslav2011, 2022-03-26:
Could you check your dongle with rtl_fm (it may require sudo due to macos hardware access restrictions)? If it does not work too, try connecting the dongle to another USB port or connecting it through a hub/docking station.

dltkddnr04, 2022-03-27:
I tried the rtl_fm command and the result is below

$ sudo rtl_fm -f 105900000         
Found 1 device(s):
  0:  , , SN: 
Using device 0: Generic RTL2832U OEM
rtlsdr_write_reg failed with -1
Resetting device...
zsh: segmentation fault  sudo rtl_fm -f 105900000

Is it possible that the SDR device failure is the cause?

vladisslav2011, 2022-03-27:
It's unlikely, that the SDR device failure is the cause.
The error code -1 means "Operation not permitted",  so there is some configuration problem. libusb can't access the device.
That's not gqrx issue as the rtl_fm have problems accessing a device too.

argilo, 2023-04-28:
It appears this issue is not due to a problem in Gqrx, so I'll close it off.

closed 2023-04-28 by argilo