gqrx-sdr/gqrx #1115: Wav recording volume too high for NFM demodulator

AsciiWolf, 2022-04-10:
Fedora 35, Gqrx 2.15.8-2 (x86_64).

When trying to record any NFM signal, the recorded file sounds distorted because the volume is obviously too high for some reason (although the actual reception in Gqrx is fine). Setting the Gain slider to a lower volume in Gqrx did not help.

[Here](https://github.com/gqrx-sdr/gqrx/files/8458926/test_recording.zip) ia an example recording. It seems to happen for all NFM signals, but WFM signals sound fine. I have tried recording the same NFM signals in another SDR software and everything was fine there.

The issue did not happen in older versions of Gqrx.

vladisslav2011, 2022-04-24:
Hello.
I have experienced this issue too.
I my case even some WFM recordings have been distorted die to high nonstandard deviation, that Gqrx can't handle well.

> Setting the Gain slider to a lower volume in Gqrx did not help.

The "gain" slider applies to audio sink only. It has nothing to do with streaming/recording.

I have fixed similar issue by replacing old audio AGC implementation with new one and putting the new AGC block into audio path regardless of demodulator selected.

Here are PR series:
https://github.com/gqrx-sdr/gqrx/pull/1056
https://github.com/gqrx-sdr/gqrx/pull/1060
https://github.com/gqrx-sdr/gqrx/pull/1061
https://github.com/gqrx-sdr/gqrx/pull/1096

You may try the latest one (as it includes 1-3 and have most bugs fixed) and give some feedback.

AsciiWolf, 2022-04-24:
Thanks for the info and links to the PRs! I will try testing it and providing feedback when I have some time.

vladisslav2011, 2022-04-24:
Hmmm.
Have you tried setting wider NFM deviation from a popup dialog:
[image]
[image]
This may be enough if you don't have mixed wide/narrow transmissions at the same channel or do not want audio normalization
.

AsciiWolf, 2022-04-24:
Thanks! Setting the max deviation to 17 kHz did work around the issue (but made weaker NFM transmissions quieter/harder to understand).

AsciiWolf, 2023-05-13:
I am not sure what changed, but the issue disappeared. It does not seem to happen anymore since Gqrx 2.15.9 (with max deviation set to 5 kHz).

AsciiWolf, 2023-05-17:
Ignore my previous comment, it still seems to be a problem when max deviation is set to 2.5 kHz.

AsciiWolf, 2023-05-22:
@argilo Could the 5 kHz max deviation of NFM demodulator be set as default in Gqrx? It still sounds good for most use cases (unlike the 17 kHz one) and makes the audio recorder work fine.

argilo, 2023-06-17:
I agree that 5 kHz would be a more sensible default for deviation, since it's commonly used in many places, and it's better for the deviation to be set too high than too low. It looks like 5 kHz was the default prior to 78fff0395c7f32ae93c406cb1552c7ea6cf280a7, and I think it's fine to put it back. I'll open a PR.

It may also make sense to add some attenuation prior to the WAV file sink so that there's some headroom to account for things like tuning offset. The default level for audio output is -6 dB and this seems to work well.

argilo, 2023-06-17:
OK, I've opened #1254 and #1255, which together should prevent most accidental clipping when making WAV file recordings.

closed 2023-06-18 by argilo