gqrx-sdr/gqrx #1116: Peak detect / Hold values not saved in config

TheRocketSurgeon, 2022-04-12:
Seems like a simple addition to save those two settings. Everything else seems to be saved on the FFT panel. :-)

```
--- dockfft.orig	2022-04-12 10:24:51.045631967 +0100
+++ dockfft.cpp	2022-04-12 10:23:41.338326527 +0100
@@ -269,6 +269,17 @@
     else
         settings->remove("bandplan");

+    // Peak
+    if (ui->peakDetectionButton->isChecked())
+        settings->setValue("peak_detect", true);
+    else
+        settings->remove("peak_detect");
+
+    if (ui->peakHoldButton->isChecked())
+        settings->setValue("peak_hold", true);
+    else
+        settings->remove("peak_hold");
+
     if (QString::compare(ui->cmapComboBox->currentData().toString(), DEFAULT_COLORMAP))
         settings->setValue("waterfall_colormap", ui->cmapComboBox->currentData().toString());
     else
@@ -345,6 +356,14 @@
     ui->bandPlanCheckbox->setChecked(bool_val);
     emit bandPlanChanged(bool_val);

+    bool_val = settings->value("peak_detect", true).toBool();
+    ui->peakDetectionButton->setChecked(bool_val);
+    emit peakDetectionToggled(bool_val);
+
+    bool_val = settings->value("peak_hold", true).toBool();
+    ui->peakHoldButton->setChecked(bool_val);
+    emit fftPeakHoldToggled(bool_val);
+
     QString cmap = settings->value("waterfall_colormap", "gqrx").toString();
     ui->cmapComboBox->setCurrentIndex(ui->cmapComboBox->findData(cmap));
```

vladisslav2011, 2022-04-24:
Hello.
Thanks for a bug report.

Good catch. I have missed this two (at least) while implementing missing settings saving/loading.
Could you open a PR?

TheRocketSurgeon, 2022-04-24:
I *think* I just did...

argilo, 2022-10-09:
This was fixed in #1121, which I've manually merged.

closed 2022-10-09 by argilo