gqrx-sdr/gqrx #1117: Bookmark and bookmark tag names don't have any text sanitization

Albert-S-Briscoe, 2022-04-16:
Bookmark tag names containing "!" (or certain other characters) can cause weird behavior, and can delete all your bookmarks. This can also happen with bookmark names, but it's less consistent.

To recreate:
1. Move your bookmarks.csv file to somewhere safe
2. Clone gqrx
3. Build with the command line or qtcreator, then start gqrx
4. Create a new bookmark (called "asdf"), and add a tag at the same time, call it "asdf"
5. In the bookmarks view, double click on the bookmark's tags to change them
6. Add a new tag called "!" or "!text" or "a!"
7. gqrx will delete all the other tags, and crash if the build was a qtcreator Debug or command line build. Sometimes the entire contents of the bookmarks.csv file is deleted.

Other problematic characters:
- Commas in bookmark tag names can expand to multiple bookmarks and/or crash gqrx when it loads the file on startup (known issue, there's a short comment about this in bookmarks.cpp)
- Semicolons (used as a field separator) have no special treatment.
- It is possible to paste newlines into the bookmark name box, which can corrupt the csv file and crash gqrx.

vladisslav2011, 2022-04-24:
Hello.
Thanks for a bug report.
What's your distribution/Qt version?

I can't confirm this in my development branch (https://github.com/vladisslav2011/gqrx/tree/fft_channelizer) on Ubuntu 18.04 Qt 5.9.5+dfsg-0ubuntu1. At least "!" character does not crash Gqrx/break bookmarks. I have started Gqrx with bookmarks.csv file removed, added some bookmarks, set some tags including "!", did several restarts, listened to some local ham communications at 2M, some railroad communications, some HDMI interference and local telemetry - no crash and no bookmarks lost.
Including ";" character into bookmark name makes this bookmark disappear and this may be fixed by proper escaping.
Including "," into tag name makes bad things (the tag gets split into multiple tags and gets added with incorrect name with comma included), but does not result in information loss. Maybe I have fixed this issue while adding more fields to bookmarks...

There are multiple possible solutions for this issue:

1. Filter out forbidden characters.
2. Implement character escaping/unescaping
3. Switch to JSON/XML

I'll test current master and update this comment.

Albert-S-Briscoe, 2022-04-25:
Hi,
Both my desktop and laptop are running Arch Linux, with Qt 6.3.0-1 and GNU Radio 3.9.5 (the official package has been at 3.9.4 for a few months, but that probably isn't relevant).

For the gqrx versions, I first found the '!' bug while working on a development branch on my desktop, but I tested it on the master branch (at 1edb983) and I got the same result. I hadn't tested it on my laptop until now... I still get the crash on the current master (1edb983), but not the official package. Note that the official package on Arch is out of date (currently at 2.15.8) and built against a different (but API compatible) version of GNU Radio, so I don't fully trust it. That's probably different on other distros.

I suspect that it's something to do with Qt, partially from when I went over the relevant configuration file code, but I haven't looked into it enough to say for sure.

I can do some more testing later today on my desktop, and at some point I can try downgrading qt.

LongnoseRob, 2022-04-25:
I also cannot reporduce it at the moment here with Gqrx v2.15.8-3-g1ef9236.

would be helpful to know what env reports for the LANG varaible on your machine, here it is LANG=en_US.UTF-8

Albert-S-Briscoe, 2022-04-25:
```
$ echo "$LANG"
en_US.UTF8
```
Nothing weird with language or locales other than my keyboard layout.

Here's the package versions of the dependencies listed on the README if it's helpful:
```
gnuradio 3.9.5.0-2 (modified)
gnuradio-iqbal 0.38.2-6
rtl-sdr 1:0.8.0-3
hackrf 2021.03.1-1
airspy 1.0.10-1
soapysdr 0.8.1-3
gnuradio-osmosdr 0.2.3-8
pulseaudio on my desktop, pipewire-pulse on my laptop
qt6-base 6.3.0-1
qt6-svg 6.3.0-1
```

derekpovah, 2022-06-07:
I would love to see support for json bookmark files. 

I can confirm this behavior with the latest release (Gqrx-2.15.9-x86_64.AppImage) on Pop!_OS 21.10.

Here's the bookmarks file that was updated with the GUI:

    # Tag name          ;  color
    Untagged            ; #c0c0c0

    # Frequency ; Name                     ; Modulation          ;  Bandwidth; Tags
      154430000; Test                     ; Narrow FM           ;       9000; Untagged
      154430000; Test;                     ; Narrow FM           ;       9000; Untagged

And the relevant output at next startup:

    $ ./Gqrx-2.15.9-x86_64.AppImage
    BookmarksFile is /home/derek/.config/gqrx/bookmarks.csv
    Bookmarks: Ignoring Line:
      154430000; Test;                     ; Narrow FM           ;       9000; Untagged